greatscottgadgets/facedancer #1: NXP LPC4330 Xplorer board firmware

yaakovgamliel, 2017-02-26:
Could you point me to the firmware needed by the NXP LPC4330 Xplorer in order to use the facedancer

ktemkin, 2017-02-28:
Sure! The LPC4330-Xplorer is supported using the GreatFET firmware, which is currently located at:

https://github.com/dominicgs/GreatFET-experimental

You can follow the _README_ in the firmware folder to get the firmware built, with one minor adjustment. When you run cmake, you'll need to specify the board you're using:

```sh
#replace "cmake .." from the README with:
cmake -DBOARD=NXP_XPLORER ..
```

You'll also need the host-side greatfet python module from that repo, which you can either install as you would any other python module or use in-place by tweaking your ```PYTHONPATH```.

Let me know how it turns out. :)

ktemkin, 2017-03-11:
I haven't heard back, so I'm assuming this was enough and closing the issue. :)

Feel free to ping back if you need help.

yaakovgamliel, 2017-03-12:
I wanted to post my results:
* Building and installing the GreatFet experimental firmware:
```bash
(py-dev) yasha@dev:~/Dev/GreatFET-experimental/firmware/greatfet_usb/build$ cmake -DBOARD=NXP_XPLORER ..
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/lib/arm-none-eabi
-- dfu-suffix not found: not building DFU file
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yasha/Dev/GreatFET-experimental/firmware/greatfet_usb/build
(py-dev) yasha@dev:~/Dev/GreatFET-experimental/firmware/greatfet_usb/build$ make
[  5%] Built target greatfet_usb_m0.elf
[  5%] Built target greatfet_usb_m0.bin
[ 94%] Built target OBJ_FILES_greatfet_usb
[ 97%] Built target greatfet_usb.elf
[ 97%] Built target greatfet_usb.bin
[100%] Built target greatfet_usb_dfu.elf
```
* Flashing the NXP LPC4330 Xplorer
```bash
(py-dev) yasha@dev:~/Dev/GreatFET-experimental/host$ ./greatfet_firmware -w ../firmware/greatfet_usb/build/greatfet_usb.bin -v
Trying to find a GreatFET device...
No GreatFET board found!
```
I also tried replacing the vendor and product id at [board.py](https://github.com/dominicgs/GreatFET-experimental/blob/master/host/greatfet/board.py) 40 with the vendor and product id from the Xplorer but it didn't work either.
Do you think I should approach the GreatFet project with this errors?
Thanks for all your help, I'm hoping to get some boards this week to build my own GreatFet, I'll post my results for both the GreatFet and the Facedancer.

ktemkin, 2017-03-13:
Looks like the documentation over in the GreatFET repository is a little bit out of date. I'll get the fixed up and post back here. :)

ktemkin, 2017-03-13:
Here's some updated documentation on programming the Xplorer-- which will hopefully be in the GreatFET repository soon:

https://github.com/ktemkin/GreatFET-experimental/blob/9f25d9ce81dc97215310e8f32691b4d71e12d373/firmware/README.md

Let me know if that helps!

yaakovgamliel, 2017-03-14:
After playing with the switches in J4 I'm having some issues to get the device into dfu, I'll check the docs for the board to find what I'm doing wrong there, thanks for all the help

yaakovgamliel, 2017-03-16:
It's working fine after following your instructions.
```bash
sudo make greatfet_usb-flash
[  5%] Built target greatfet_usb_m0.elf
[  5%] Built target greatfet_usb_m0.bin
[ 97%] Built target OBJ_FILES_greatfet_usb
[100%] Built target greatfet_usb.elf
[100%] Built target greatfet_usb.bin
[100%] Built target greatfet_usb-flash
``` 
thank you for all your help!

closed 2017-03-16 by yaakovgamliel