greatscottgadgets/facedancer #8: Proxing a keyboard

ikarus23, 2018-10-07:
Hi,
I'm trying to proxy a keyboard just for fun and learning. While the Nintendo Pro Controller works great, the keyboard does not. It looks like something is not reacting right to the response of the get device descriptor request. It times out.

```
sudo ./facedancer-usbproxy.py -v 04d9 -p 1603
NOTE: Skipping GoodFET-based devices, as pyserial isn't installed.
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
[14:36:57] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[14:36:57] <: b'\x12\x01\x10\x01\x00\x00\x00\x08\xd9\x04\x03\x16\x10\x03\x01\x02\x00\x01'
[14:37:08] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[14:37:08] <: b'\x12\x01\x10\x01\x00\x00\x00\x08\xd9\x04\x03\x16\x10\x03\x01\x02\x00\x01'
Traceback (most recent call last):
  File "./facedancer-usbproxy.py", line 50, in <module>
    main()
  File "./facedancer-usbproxy.py", line 44, in main
    d.run()
  File "/home/user/Facedancer/facedancer/USBDevice.py", line 164, in run
    self.scheduler.run()
  File "/home/user/Facedancer/facedancer/core.py", line 492, in run
    task()
  File "/home/user/Facedancer/facedancer/USBDevice.py", line 84, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 767, in service_irqs
    self._handle_setup_events()
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 368, in _handle_setup_events
    self._handle_setup_event_on_endpoint(i)
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 406, in _handle_setup_event_on_endpoint
    self.ack_status_stage(direction=self.DEVICE_TO_HOST)
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 145, in ack_status_stage
    self.read_from_endpoint(endpoint_number)
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 269, in read_from_endpoint
    while not self._transfer_is_complete(ep_num, self.HOST_TO_DEVICE):
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 429, in _transfer_is_complete
    status = self._fetch_transfer_status()
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 416, in _fetch_transfer_status
    return self._fetch_status_register(self.GET_ENDPTCOMPLETE)
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 328, in _fetch_status_register
    raw_status = self.device.vendor_request_in(self.vendor_requests.GREATDANCER_GET_STATUS, index=register_number, length=4)
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/board.py", line 396, in vendor_request_in
    value=value, index=index, timeout=timeout)
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/board.py", line 384, in _vendor_request
    request, value, index, length_or_data, timeout)
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg/usb/core.py", line 1043, in ctrl_transfer
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg/usb/backend/libusb1.py", line 883, in ctrl_transfer
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg/usb/backend/libusb1.py", line 595, in _check
usb.core.USBError: [Errno 110] Operation timed out
```
Any ideas?

dominicgs, 2018-10-08:
I think this is related to you running the latest firmware and the new GreatFET USB API not being ready for FaceDancer yet.  Can I suggest that you roll back your GreatFET firmware and host code to commit 72d2b3111e6426b2ea80135f5536514eff8edbcf for the time being?

ikarus23, 2018-10-08:
I reverted to https://github.com/greatscottgadgets/greatfet/commit/72d2b3111e6426b2ea80135f5536514eff8edbcf but the result is still the same.

I've also tested it with another keyboard, but the result is still the same. Also interesting: after the crash the GreatFET is unresponsive (`DeviceNotFoundError`). Sometimes, there isen't even a crash. It just stays at this forever (causing "high" CPU load):
```
sudo ./facedancer-usbproxy.py -v 046a -p 0023                                                                               :(
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
[22:42:54] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[22:42:54] <: b'\x12\x01\x00\x02\x00\x00\x00\x08j\x04#\x00 \x02\x00\x00\x00\x01'
[22:43:01] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[22:43:01] <: b'\x12\x01\x00\x02\x00\x00\x00\x08j\x04#\x00 \x02\x00\x00\x00\x01'
-- Reset requested! --
```

dominicgs, 2018-10-09:
From the looks of the original traceback, it's having an issue talking to the GreatFET, not the keyboard.  Which is why I thought it might be related to the USB API for the GreatFET.

ikarus23, 2018-10-09:
Well, in most cases the error is the same. Just sometimes the "hang" occurs. There are even cases where the error is `[Errno 5] Input/Output Error` (vs. `[Errno 110] Operation timed out`). It's not very deterministic...

```
sudo ./facedancer-usbproxy.py -v 046a -p 0023
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
[22:42:54] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[22:42:54] <: b'\x12\x01\x00\x02\x00\x00\x00\x08j\x04#\x00 \x02\x00\x00\x00\x01'
[22:43:01] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[22:43:01] <: b'\x12\x01\x00\x02\x00\x00\x00\x08j\x04#\x00 \x02\x00\x00\x00\x01'
-- Reset requested! --
Traceback (most recent call last):
  File "./facedancer-usbproxy.py", line 50, in <module>
    main()
  File "./facedancer-usbproxy.py", line 44, in main
    d.run()
  File "/home/user/Facedancer/facedancer/USBDevice.py", line 164, in run
    self.scheduler.run()
  File "/home/user/Facedancer/facedancer/core.py", line 492, in run
    task()
  File "/home/user/Facedancer/facedancer/USBDevice.py", line 84, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 757, in service_irqs
    status = self._fetch_irq_status()
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 339, in _fetch_irq_status
    return self._fetch_status_register(self.GET_USBSTS)
  File "/home/user/Facedancer/facedancer/backends/GreatDancerApp.py", line 328, in _fetch_status_register
    raw_status = self.device.vendor_request_in(self.vendor_requests.GREATDANCER_GET_STATUS, index=register_number, length=4)
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/board.py", line 379, in vendor_request_in
    value=value, index=index, timeout=timeout)
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/GreatFET-0.0-py3.7.egg/greatfet/board.py", line 367, in _vendor_request
    request, value, index, length_or_data, timeout)
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg/usb/core.py", line 1043, in ctrl_transfer
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg/usb/backend/libusb1.py", line 883, in ctrl_transfer
  File "/home/user/gf-fd-venv/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg/usb/backend/libusb1.py", line 595, in _check
usb.core.USBError: [Errno 5] Input/Output Error
```

ktemkin, 2018-10-10:
I'd have to take a closer look with a keyboard of my own-- but I suspect
this might be an issue with the way we're trying to proxy a Low Speed
device through a Full Speed connection.

There's currently an an issue in the GreatFET USB driver that causes
transactions after a failed transaction to fail. I have it fixed (and a
bunch of other USB stability improvements) in one of my branches--
I'll work to start the merge process for those tomorrow. Those improvements
should give us more of an ability to diagnose what's going on. I'd suggest
hanging tight until those are in; we can check out the issue more after.

--Kate

xairy, 2018-10-10:
I'm having the same issue with the NXP LPC4330 Xplorer board while trying to proxy a mouse:

```
$ sudo ./facedancer-usbproxy.py -v 046d -p c077
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
[22:50:10] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[22:50:10] <: b'\x12\x01\x00\x02\x00\x00\x00\x08m\x04w\xc0\x00r\x01\x02\x00\x01'
[22:50:15] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[22:50:15] <: b'\x12\x01\x00\x02\x00\x00\x00\x08m\x04w\xc0\x00r\x01\x02\x00\x01'
-- Reset requested! --
-- Reset requested! --
[22:50:15] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[22:50:15] <: b'\x12\x01\x00\x02\x00\x00\x00\x08m\x04w\xc0\x00r\x01\x02\x00\x01'
Traceback (most recent call last):
  File "./facedancer-usbproxy.py", line 50, in <module>
    main()
  File "./facedancer-usbproxy.py", line 44, in main
    d.run()
  File "/home/user/Tools/usb/Facedancer/facedancer/USBDevice.py", line 164, in run
    self.scheduler.run()
  File "/home/user/Tools/usb/Facedancer/facedancer/core.py", line 139, in run
    task()
  File "/home/user/Tools/usb/Facedancer/facedancer/USBDevice.py", line 84, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 767, in service_irqs
    self._handle_setup_events()
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 368, in _handle_setup_events
    self._handle_setup_event_on_endpoint(i)
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 403, in _handle_setup_event_on_endpoint
    self.connected_device.handle_request(request)
  File "/home/user/Tools/usb/Facedancer/facedancer/USBProxy.py", line 220, in handle_request
    self._proxy_in_request(req)
  File "/home/user/Tools/usb/Facedancer/facedancer/USBProxy.py", line 265, in _proxy_in_request
    self.send_control_message(data)
  File "/home/user/Tools/usb/Facedancer/facedancer/USBDevice.py", line 196, in send_control_message
    self.maxusb_app.send_on_endpoint(0, data)
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 252, in send_on_endpoint
    while not self._transfer_is_complete(ep_num, self.DEVICE_TO_HOST):
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 429, in _transfer_is_complete
    status = self._fetch_transfer_status()
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 416, in _fetch_transfer_status
    return self._fetch_status_register(self.GET_ENDPTCOMPLETE)
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 328, in _fetch_status_register
    raw_status = self.device.vendor_request_in(self.vendor_requests.GREATDANCER_GET_STATUS, index=register_number, length=4)
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-0.0-py3.5.egg/greatfet/board.py", line 379, in vendor_request_in
    value=value, index=index, timeout=timeout)
  File "/usr/local/lib/python3.5/dist-packages/GreatFET-0.0-py3.5.egg/greatfet/board.py", line 367, in _vendor_request
    request, value, index, length_or_data, timeout)
  File "/home/user/.local/lib/python3.5/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/home/user/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/home/user/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
```

At the same time just emulating a keyboard works perfectly:

```
$ sudo ./facedancer-keyboard.py 
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device USB keyboard device
-- Reset requested! --
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
USB keyboard device received GET_DESCRIPTOR req 1, index 0, language 0x0000, length 64
[... 7 lines cut ...]
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=DEVICE_QUALIFIER descriptor (index=0x00), index=0, length=10)
USB keyboard device received GET_DESCRIPTOR req 6, index 0, language 0x0000, length 10
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
USB keyboard device received GET_DESCRIPTOR req 2, index 0, language 0x0000, length 9
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=34)
USB keyboard device received GET_DESCRIPTOR req 2, index 0, language 0x0000, length 34
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x00), index=0, length=255)
USB keyboard device received GET_DESCRIPTOR req 3, index 0, language 0x0000, length 255
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=255)
[... cut ...]
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
USB keyboard device received GET_DESCRIPTOR req 3, index 3, language 0x0409, length 255
USB keyboard device received request >, class request to interface (class request 10: value=0, index=0, length=0)
USB keyboard device invalid handler entity, stalling: dir=0, type=1, rec=1, r=a, v=0, i=0, l=0
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=REPORT descriptor (index=0x00), index=0, length=43)
```

ikarus23, 2019-03-24:
Some time has passed and I though I'd give it another try. I used the latest code from Greatfet/Facedancer to proxy the ACR122u NFC reader (just for fun). The result is different, but it still does not work.
```
> gf info                           
Found a GreatFET One!                                               
  Board ID: 0
  Firmware version: git-46ff6e3
  Part ID: a0000a306c4753
  Serial number: 0000d1c466e635143c13

>  python facedancer-usbproxy.py -v 072f -p 2200
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
-- Reset requested! --
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
-- Reset requested! --
-- Reset requested! --
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
[19:52:28] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:28] < --STALLED-- 
-- Reset requested! --
[19:52:29] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:29] < --STALLED-- 
[19:52:29] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:29] < --STALLED-- 
[19:52:29] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[19:52:29] < --STALLED-- 
-- Reset requested! --
-- Reset requested! --
-- Reset requested! --
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
-- Reset requested! --
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
[19:52:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[19:52:30] < --STALLED-- 
```
After this it just hangs.

ktemkin, 2019-03-24:
This looks a lot like your current user doesn't have the right permissions to be able to talk to the NFC reader's device node. If you're willing, can you try running the command again with sudo?

`sudo python facedancer-usbproxy.py -v 072f -p 2200`

(Or, if you'd prefer to avoid sudo'ing the script, and you're on linux, you can likely change the `usbfs` node's permissions to be rw for your user.)

ikarus23, 2019-03-26:
Thanks for the feedback. So I ran facedancer as root (like I do with all code I checkout on github :) ) and some things changed. Now I only get:
```
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
-- Reset requested! --
[22:37:41] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[22:37:41] <: b'\x12\x01\x10\x01\x00\x00\x00\x08/\x07\x00"\x14\x02\x01\x02\x00\x01'
```
And that's it, nothing more. However, the NFC reader (the target device) is still working.

amado4884, 2019-06-24:
I'm having the same issue. It just hangs there. I haven't been able to get anything working an I'm worried that maybe I've got a faulty GreatFet.

ktemkin, 2019-07-04:
This looks a lot like a software issue to me -- USBProxy hasn't been validated against low-speed devices; and I suspect the issue is related to it attempting to proxy data from a low-speed device using the FaceDancer high speed connection.

I'm finally getting a chance to look into this, so I'll get back to you shortly with an update. :)

ktemkin, 2019-07-13:
This turned out to be a really fun one. The short version is that the Linux kernel violates the USB specification in a way I didn't expect:

https://github.com/usb-tools/Facedancer/blob/master/facedancer/USBProxy.py#L191

There's a workaround for Linux's er, liberties in master, now. (Interrupt endpoints may not have the best performance with some low-speed devices, but this is a separate issue I'll get to eventually.)

ktemkin, 2019-07-13:
Can you test this out and let me know if it works for you? Thanks!

ikarus23, 2019-07-15:
Yeeees! Thank you so much! I can now proxy my USB 1.1 Keyboard (HID device).

closed 2019-07-15 by ikarus23

ikarus23, 2019-07-15:
However, proxying  a USB 2.0 and 3.0 flash drive did not work. At least not completely. I think there is some kind of read error. I will open another issue with more details after I did some more testing.