greatscottgadgets/facedancer #13: NXP Xplorer LPC4330 alternative

coley7, 2019-02-27:
Because this board will not be produced anymore, is there another alternative available? I mean an existing evalutation kit.

ktemkin, 2019-03-04:
I don't know of an existing NXP evaluation kit with *direct* compatibility; though there are a few general options:

- The @greatscottgadgets GreatFET One / GreatFET Azalea should be available Very Soon (TM); at which point it will be the "canonical" FaceDancer2 board. 
- A few of the LPCXpresso boards have mostly-compatible microcontrollers; if someone wanted to add support for their uC-variants to the GreatFET software distribution ([greatscottgadgets/greatfet](https://www.github.com/greatscottgadgets/greatfet)), that'd be an easy way to maintain support.

Does that answer your question?

coley7, 2019-03-10:
Thank you very much for the information.

closed 2019-03-10 by coley7