greatscottgadgets/facedancer #18: facedancer-serial: ValueError and IndexError

nepheme, 2019-07-26:
Hi!
I'm using the Facedancer21 from int3.cc and I'm trying to run facedancer-serial.py, but it only starts working after 10 or 11 failed attempts every time I plug the board in. 

The first few commands (not sure if they even are commands) I'm getting don't seem to follow the structure in `readcmd()`, so `len(data) != n` is always true and raises a `ValueError`. 
The last command just before it works is identical with the one when it works, but in the first case the next command I'm reading is empty and I'm getting an `Index Error`.

Could this be an issue with the board itself?
The only information I could find about this was someone having the same problem but with a completely different script (https://github.com/nccgroup/umap/issues/11), but that didn't help much either.

Here are the errors I'm getting when it doesn't work:
(including what was read in `read(self, n)`) 
```
Using MAXUSB backend.
b'80bf8a80'
b'b4ba3a584de969f7f5d1c9c9e8aeb3b297b759bdf9'
Traceback (most recent call last):
  File "./facedancer-serial.py", line 8, in <module>
    u = FacedancerUSBApp(verbose=1)
  File "/root/Desktop/Facedancer/facedancer/core.py", line 23, in FacedancerUSBApp
    return FacedancerApp.autodetect(verbose, quirks)
  File "/root/Desktop/Facedancer/facedancer/core.py", line 54, in autodetect
    return subclass(verbose=verbose, quirks=quirks)
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 45, in __init__
    device = Facedancer(serial, verbose=verbose)
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 146, in __init__
    self.reset()
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 163, in reset
    c = self.readcmd()
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 204, in readcmd
    + ' bytes but received only ' + str(len(data)))
ValueError: Facedancer expected 32906 bytes but received only 21

```

The last time it doesn't work I get this error:
```
Using MAXUSB backend.
b'007f1600'
b'687474703a2f2f676f6f646665742e73662e6e65742f'
Facedancer reset
GoodFET monitor initialized
b''
Traceback (most recent call last):
  File "./facedancer-serial.py", line 8, in <module>
    u = FacedancerUSBApp(verbose=1)
  File "/root/Desktop/Facedancer/facedancer/core.py", line 23, in FacedancerUSBApp
    return FacedancerApp.autodetect(verbose, quirks)
  File "/root/Desktop/Facedancer/facedancer/core.py", line 54, in autodetect
    return subclass(verbose=verbose, quirks=quirks)
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 45, in __init__
    device = Facedancer(serial, verbose=verbose)
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 148, in __init__
    self.monitor_app.announce_connected()
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 391, in announce_connected
    resp = self.device.readcmd()
  File "/root/Desktop/Facedancer/facedancer/backends/GoodFETMaxUSBApp.py", line 193, in readcmd
    app = b[0]
IndexError: index out of range
```

The number of expected bytes changes for each time I run the script:
```
b'80bf8a80'
b'b4ba3a584de969f7f5d1c9c9e8aeb3b297b759bdf9'
...
ValueError: Facedancer expected 32906 bytes but received only 21

b'80bf8a80'
b'b4ba5a584ee979f5bdc1cda5b496aed6c953453dff'
...
ValueError: Facedancer expected 32906 bytes but received only 21

b'80bf8b80'
b'b45a5e5ce7e979f5fdc9cda5bc97aed6c953457dff'
...
ValueError: Facedancer expected 32907 bytes but received only 21

b'80bf4b40'
b'5a2d174bbabddddfbfb273292e29c95caeb27af9'
...
ValueError: Facedancer expected 16459 bytes but received only 20

b'80bf4b40'
b'2a87e19a9779f5ddb4b3292e39c95cbeb25af9'
...
ValueError: Facedancer expected 16459 bytes but received only 19

b'80298074'
b'5d2d8ed3f19addb7725929cf31c9967729e7f9'
...
ValueError: Facedancer expected 29824 bytes but received only 19

b'80298054'
b'5d2f96ba5db7f3c9b629cf31c9965729eff9'
...
ValueError: Facedancer expected 21632 bytes but received only 18

b'00feb104'
b'106dbbf4e0e9f2e9d1aed3132eaed9765ecd3171e22b2eddf1'
...
ValueError: Facedancer expected 1201 bytes but received only 25

b'00ff2600'
b'd8e4e4f07a4f5fc7dfcfc4c6c5e45ee3c65ecec5e44f'
...
ValueError: Facedancer expected 38 bytes but received only 22

b'007f1600'
b'687474703a2f2f676f6f646665742e73662e6e65742f'
Facedancer reset
GoodFET monitor initialized
b''
...
IndexError: index out of range

Using MAXUSB backend.
b'007f1600'
b'687474703a2f2f676f6f646665742e73662e6e65742f'
Facedancer reset
GoodFET monitor initialized
b'00b10000'
MAXUSB initialized
b'40100000'
b'40100000'
... // works

```

keymandll, 2020-07-13:
Same here with `goodfet` backend.

straithe, 2024-08-15:
@nepheme do you still have this issue?

straithe, 2024-09-19:
We are trying to find a Facedancer21 to validate this issue.