greatscottgadgets/facedancer #19: Explicitly claim the interface for the GreatDancer backend

Qyriad, 2019-08-22:
As of https://github.com/greatscottgadgets/libgreat/commit/ee45e39c5a151a68cbd9f68c04307948336acecf, GreatDancer needs to be updated to explicitly claim the USB interface so that performance isn't terrible.

Currently, udev on Linux gets saturated by the repeated binds and unbinds.

bhass1, 2020-02-14:
This is making my Ubuntu 18.04 laptop really angry.

After killing whatever script is using the GreatDancer backend, I need to issue the following so my laptop chills out:

```
sudo systemctl restart systemd-udevd systemd-udevd-kernel.socket systemd-udevd-control.socket
```

Unfortunately the linked reference fix doesn't seem trivial to apply here...

closed 2020-02-20 by Qyriad

ktemkin, 2020-02-20:
@bhass1 Can you try the most recent `master` and see if that calms things down for you? 

There was a kernel change that broke this part of userspace a bit ago. :(

bhass1, 2020-02-21:
@ktemkin & @Qyriad - But the kernel never breaks userspace! /s 

The workaround helps! I still get bind/unbind churn while exclusive access is held, but at least when the script is terminated and exclusive access is released, there isn't any more churn. Thank you!

commit e3a09fde (2020-02-20): GreatDancer: use exclusive access to keep udev happy on Linux

fixes #19