greatscottgadgets/facedancer #20: Proxying a USB flash drive

ikarus23, 2019-08-23:
Hi Kate, I'm trying to proxy a simple USB 2.0 flash drive. There is no error in the output of Facedancer but the device never shows up for mounting and in `dmesg` there are some `device descriptor read/64, error -71` entries. Any ideas?

decidedlygray, 2019-12-02:
@ikarus23 were you ever able to get this working? using the latest GreatFET One sw/fw. I am getting a crash in Facedancer though, so it may be a different issue

ikarus23, 2019-12-23:
Hi @decidedlygray. No it is not working. With the latest firmware the whole setup breaks with a new error.
```
sudo python facedancer-usbproxy.py -v 4146 -p ba65
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
-- Reset requested! --
Traceback (most recent call last):
  File "/home/ikarus/Documents/hack/usb/greatfet-venv/lib/python3.8/site-packages/pygreat/comms.py", line 780, in execute_command
    result = self.unpack(out_format, raw_result)
  File "/home/ikarus/Documents/hack/usb/greatfet-venv/lib/python3.8/site-packages/pygreat/comms.py", line 617, in unpack
    bytes_consumed, raw_bytes = cls._split_off_bytes_for_format(subformat, raw_bytes)
  File "/home/ikarus/Documents/hack/usb/greatfet-venv/lib/python3.8/site-packages/pygreat/comms.py", line 425, in _split_off_bytes_for_format
    num_bytes_consumed = cls._get_bytes_consumed_by_format(format_string, raw_bytes)
  File "/home/ikarus/Documents/hack/usb/greatfet-venv/lib/python3.8/site-packages/pygreat/comms.py", line 402, in _get_bytes_consumed_by_format
    return int(match.groups(1))
TypeError: int() argument must be a string, a bytes-like object or a number, not 'tuple'
```

decidedlygray, 2019-12-23:
@ikarus23 here is a workaround for that error message: https://github.com/greatscottgadgets/libgreat/issues/18

TLBarber, 2020-01-27:
I'm also having an issue with this but I don't get the error until after the device has been enumerated, I'm assuming it's when the target tries to mount the device.
`sudo ./facedancer-usbproxy.py -v 0781 -p 5530
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device Proxy'd USB Device
-- Reset requested! --
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[10:47:50] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x070U\x00\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
-- Reset requested! --
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[10:47:50] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x070U\x00\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=DEVICE_QUALIFIER descriptor (index=0x00), index=0, length=10)
[10:47:50] <: b'\n\x06\x00\x02\x00\x00\x00@\x01\x00'
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
[10:47:50] <: b'\t\x02 \x00\x01\x01\x00\x80d'
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=32)
[10:47:50] <: b'\t\x02 \x00\x01\x01\x00\x80d\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x02\x02\x00\x02\x01'
-- Storing configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=200mA> --
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x00), index=0, length=255)
[10:47:50] <: ̄Љ
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=255)
[10:47:50] <: ̎Cruzer
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x01), index=409, length=255)
[10:47:50] <: ̐SanDisk
[10:47:50] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
[10:47:50] <: ̪20054963600A4E2343C3
[10:47:50] >, standard request to device (SET_CONFIGURATION: value=1, index=0, length=0)
-- Applying configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=200mA> --
Setting up endpoint 1 (direction=1, transfer_type=2, max_packet_size=512)
Setting up endpoint 2 (direction=0, transfer_type=2, max_packet_size=512)
[10:47:51] <, class request to interface (class request 254: value=0, index=0, length=1)
[10:47:51] <: b'\x00'
[10:47:51] 2>: b'USBC\x01\x00\x00\x00$\x00\x00\x00\x80\x00\x06\x12\x00\x00\x00$\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
[10:48:12] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[10:48:12] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x070U\x00\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
-- Reset requested! --
[10:48:13] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[10:48:13] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x070U\x00\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
-- Reset requested! --
-- Reset requested! --
[10:48:13] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[10:48:13] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x070U\x00\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
-- Reset requested! --
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[10:48:14] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x070U\x00\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=DEVICE_QUALIFIER descriptor (index=0x00), index=0, length=10)
[10:48:14] <: b'\n\x06\x00\x02\x00\x00\x00@\x01\x00'
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
[10:48:14] <: b'\t\x02 \x00\x01\x01\x00\x80d'
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=32)
[10:48:14] <: b'\t\x02 \x00\x01\x01\x00\x80d\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x02\x02\x00\x02\x01'
-- Storing configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=200mA> --
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x00), index=0, length=255)
[10:48:14] <: ̄Љ
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=255)
[10:48:14] <: ̎Cruzer
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x01), index=409, length=255)
[10:48:14] <: ̐SanDisk
[10:48:14] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
[10:48:14] <: ̪20054963600A4E2343C3
[10:48:14] >, standard request to device (SET_CONFIGURATION: value=1, index=0, length=0)
-- Applying configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=200mA> --
Setting up endpoint 1 (direction=1, transfer_type=2, max_packet_size=512)
Setting up endpoint 2 (direction=0, transfer_type=2, max_packet_size=512)
[... 20 lines cut ...]
    self.scheduler.run()
  File "/home/cyberbabayaga/Facedancer/facedancer/core.py", line 506, in run
    task()
[... cut ...]
    data = self.libusb_device.read(endpoint_address, endpoint.max_packet_size, timeout=timeout)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
`

straithe, 2021-11-13:
Hi, @ikarus23 are you still experiencing this issue?

closed 2021-12-12 by straithe