greatscottgadgets/facedancer #24: can't list usb emulated device programmatically on macOS

giorgio-zamparelli, 2019-10-01:
An USB device emulated with Facedancer and GreatFET One will not be listed by libusb or [IOKit](https://stackoverflow.com/questions/7567872/how-to-create-a-program-to-list-all-the-usb-devices-in-a-mac) on macOS even though is shown by CLI commands [lsusb](https://github.com/jlhonora/lsusb) / [system_profiler SPUSBDataType](https://apple.stackexchange.com/questions/170105/list-usb-devices-on-osx-command-line)

STEPS TO REPRODUCE
Connect a GreatFET One to both ports on a MacBook.
Run ```python3 ./facedancer-keyboard.py```
Run ```lsusb``` or ```system_profiler SPUSBDataType``` and you should see listed both the GreatFET One and the emulated USB keyboard:
<pre><code>Bus 020 Device 004: ID 05ac:0274 Apple Inc. Apple Internal Keyboard / Trackpad  Serial: D3H54562C61FTV4AX1FS
Bus 020 Device 005: ID 05ac:8290 Apple Inc. Bluetooth USB Host Controller 
Bus 020 Device 000: ID 610b:4653 610b MAX3420E Enum Code  Serial: S/N3420E
Bus 020 Device 021: ID 1d50:60e6 1d50 GreatFET  Serial: 000000000000000057cc67e631923157</code></pre>
List of USB devices programmatically using either libusb, [pyusb](https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst) or IOKit:
<pre><code>import usb.core
devices = usb.core.find(find_all=True)
for device in devices:
  print("device", hex(device.idVendor), ":", hex(device.idProduct))</code></pre>
the emulated usb device will not be detected even though the GreatFET One is detected.
<pre><code>device 0x1d50 : 0x60e6 // GreatFET
device 0x5ac : 0x8290 // Apple Inc. Bluetooth USB Host Controller
device 0x5ac : 0x274 // Apple Inc. Apple Internal Keyboard / Trackpad
device 0x5ac : 0x8007 // ???</code></pre>

mrstock said on [Discord](https://discord.gg/HKAhHub) that he also couldn't list an emulated USB device programmatically on macOS but it could do so on Ubuntu.

mrstock said that on macOS the descriptor reading phase is not performed and the endpoints are not set. It remains in an infinite loop waiting for a transfer to complete on [line 243 of GreatDancerApp.py](https://github.com/usb-tools/Facedancer/blob/cf52818eedb611e5cb714164bf1551cc6648913b/facedancer/backends/GreatDancerApp.py#L243)
<pre><code>USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=18)
USB keyboard device received GET_DESCRIPTOR req 3, index 3, language 0x0409, length 18
[18, 3, 83, 0, 47, 0, 78, 0, 51, 0, 52, 0, 50, 0, 48, 0, 69, 0]
sending on 0: bytearray(b'\x12\x03S\x00/\x00N\x003\x004\x002\x000\x00E\x00')
Cleaning up transfers on 0
Out status: 0b1
IN status: 0b0
Cleaning up transfers on 0
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
USB keyboard device received GET_DESCRIPTOR req 2, index 0, language 0x0000, length 9
[9, 2, 34, 0, 1, 1, 4, 224, 250]
sending on 0: b'\t\x02"\x00\x01\x01\x04\xe0\xfa'
</code></pre>
it stops alway at this point

System Info
OS: macOS High Sierra 10.13.6
Hardware: MacBook Pro (Retina, 15-inch, Mid 2015)
GreatFET One:
  Board ID: 0
  Firmware version: v2019.9.1
  Part ID: a0000a30624362
  Serial number: 000057cc67e631923157

Python 2.7.13
pyusb 1.0.2

[image]

[image]

straithe, 2021-11-13:
Hi, @giorgio-zamparelli. Are you still experiencing this issue?

giorgio-zamparelli, 2021-11-14:
In the end I installed VMWare Fusion on my MacBook with Ubuntu and used the GreatFET One from Ubuntu.
It would work also with other free virtual machine hypervisors like VirtualBox:
https://www.virtualbox.org/

straithe, 2021-11-14:
Thank you for the update! I'm going to close this issue now as you seem to have solved the issue you were experiencing.

closed 2021-11-14 by straithe