greatscottgadgets/facedancer #31: Dealing with USB descriptor longer than declared length

HclX, 2020-01-31:
I'm running into parsing issues with a descriptor blob longer than specification says:

[image]

This is from a USB headset (Plantronics C5220T). The above snapshot is from wireshark capture of the device's CONFIGURATION descriptor. As it shows, the device is reporting a 9-byte endpoint descriptor, rather than a 7-byte according to the USB specification. 

I was trying to parse it using facedancer's USBEndpoint.from_binary_descriptor class and it fails at "struct.unpack" call.

ktemkin, 2020-01-31:
Sounds like there needs to be a `[0:7]` on the data passed to `struct.unpack`;  or we should use the `wLength` reported in the descriptor itself.

HclX, 2020-01-31:
The same should apply to all the "USBDescribable" subclasses. If needed, I can make a pull request for this.

ktemkin, 2020-01-31:
Agreed; and if you'd like to, I'd be happy to test and merge that. :)

Thanks!

ktemkin, 2020-02-20:
This should have been resolved by https://github.com/usb-tools/Facedancer/pull/32; closing.

closed 2020-02-20 by ktemkin