greatscottgadgets/facedancer #34: GreatFET One  FaceDancer  USB Proxy  How to MITM?

Eddiiie, 2020-02-20:
Hi,
I got a beagle bone black to run USB Proxy on a device I am trying to understand.   Eventually I got USB Proxy-Legacy to run, and the host device fails to run the software...  Some packets are exchanged and program stops.  I believe I saw something about too much latency (100ms)..

So I buy GreatFET, Amazon delivery, here the next day, cool.     Gave up on Windows, moved to Raspberry Pi, used Ubuntu, has some issues, third try use Raspian ..  Works right off the bat!  Nice.

The paragraph I find for help (read.me)  for Facedancer says   export BACKEND = greatfet    ./facedancer-serial.py   (which runs).      Looking in the directory, I see some other facedancer-usb* files,   what do I run to put it in MITM mode?   Is there a wiki page somewhere?

Also, does anyone else think facesitter instead of facedancer?  Does that make me a bad person?

ktemkin, 2020-02-20:
>  Looking in the directory, I see some other facedancer-usb* files, what do I run to put it in MITM mode? Is there a wiki page somewhere?

The scripts that contain `usbproxy` in the name are the MITM scripts; the core one is `facedancer-usbproxy`. There's a diagram in the README of the hookup you'll want for that. :)

> facedancer-usbproxy.py wants -v and -p. Do I have to know this? I did not have to pass that on the legacy version.

`-v` and `-p` are the vendor and product ID of the device you're going to pass through. The old USBProxy only had one USB port, so it didn't need help figuring out which device you were working with. Your PC potentially has quite a few, so it needs the disambiguation. :)

> Also, does anyone else think facesitter instead of facedancer? Does that make me a bad person?

Nope; but I'm certainly not judging.~

closed 2020-02-20 by ktemkin

Eddiiie, 2020-02-22:
Hello @ktemkin,

Pleasure to meet you.    Forgive my ignorance, the ASCII picture in the Readme is very nice, but I am still not clear.  :)  GreatFET has 2 USB ports,  one I am using to power and talk to the GreatFET.  The other port goes to the device I want to read from (let's say Joystick).   How do I then connect to the GreatFET to the host computer's USB port (say game console)? 

I have no access to the Target Host's OS.

When I plug in the device I want to read from (say joystick), there is no power to the device.

Thank you for your help!!