greatscottgadgets/facedancer #36: facedancer-keyboard.py broken

xairy, 2020-02-25:
Commit 59425336fd6486cea457d60f15e2336bcd64b498 broke `facedancer-keyboard.py`. Reverting the commit fixes the issue.

``` bash
# ./facedancer-keyboard.py 
Using GreatDancer backend.
GreatDancer initialized
GreatDancer connected device USB keyboard device
-- Reset requested! --
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
USB keyboard device received GET_DESCRIPTOR req 1, index 0, language 0x0000, length 64
-- Reset requested! --
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
USB keyboard device received GET_DESCRIPTOR req 1, index 0, language 0x0000, length 18
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=34)
USB keyboard device received GET_DESCRIPTOR req 2, index 0, language 0x0000, length 34
USB keyboard device received request <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
USB keyboard device received GET_DESCRIPTOR req 3, index 3, language 0x0409, length 255
USB keyboard device received request >, standard request to device (SET_CONFIGURATION: value=1, index=0, length=0)
USB keyboard device received SET_CONFIGURATION request
Setting up endpoint 3 (direction=1, transfer_type=3, max_packet_size=512)
USB keyboard interface sending keypress 0x00
USB keyboard device received request <, standard request to interface (GET_DESCRIPTOR: value=REPORT descriptor (index=0x00), index=0, length=43)
USB keyboard interface received GET_DESCRIPTOR at interface req 34, index 0, language 0x0000, length 43
USB keyboard device received request >, class request to interface (class request 10: value=0, index=0, length=0)
Traceback (most recent call last):
  File "./facedancer-keyboard.py", line 14, in <module>
    d.run()
  File "/home/user/Tools/usb/Facedancer/facedancer/USBDevice.py", line 166, in run
    self.scheduler.run()
  File "/home/user/Tools/usb/Facedancer/facedancer/core.py", line 506, in run
    task()
  File "/home/user/Tools/usb/Facedancer/facedancer/USBDevice.py", line 84, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 760, in service_irqs
    self._handle_setup_events()
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 367, in _handle_setup_events
    self._handle_setup_event_on_endpoint(i)
  File "/home/user/Tools/usb/Facedancer/facedancer/backends/GreatDancerApp.py", line 401, in _handle_setup_event_on_endpoint
    self.connected_device.handle_request(request)
  File "/home/user/Tools/usb/Facedancer/facedancer/USBDevice.py", line 236, in handle_request
    handler_entity = recipient.interface
AttributeError: 'USBKeyboardInterface' object has no attribute 'interface'
```

ktemkin, 2020-02-29:
Yep, I agree there's a regression and agree on the source. Should fixed shortly.

xairy, 2020-03-30:
This is now fixed.

closed 2020-03-30 by xairy