greatscottgadgets/facedancer #41: facedancer-usbproxy.py - trying to sniff a usb audio device

MrBollie, 2020-03-26:
Hello,

I'm trying to use the proxy-script and a GreatFET to proxy and sniff an RME BabyFace Pro usb audio device in class compliant mode like this

(babyface pro) - (host computer) - (greatFETOne) - (ipad with camera connection kit)

lsusb:
`Bus 001 Device 015: ID 2a39:3fb0`

so I call:
`./facedancer-usbproxy.py -v 2a39 -p 3fb0`

resulting in:
`Using GreatDancer backend.
GreatDancer initialized
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[14:19:06] <: b'\x12\x01\x00\x02\xef\x02\x01@9*\xb0?\x01\x00\x01\x02\x03\x01'
-- Patched device descriptor. --
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[14:19:06] <: b'\x12\x01\x00\x02\xef\x02\x01@9*\xb0?\x01\x00\x01\x02\x03\x01'
-- Patched device descriptor. --
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=DEVICE_QUALIFIER descriptor (index=0x00), index=0, length=10)
[14:19:06] <: b'\n\x06\x00\x02\xef\x02\x01@\x00\x00'
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
[14:19:06] <: b'\t\x02\xa7\x01\x04\x01\x00\x802'
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=423)
[14:19:06] <: b'\t\x02\xa7\x01\x04\x01\x00\x802\x08\x0b\x00\x04\x01\x00 \x00\t\x04\x00\x00\x00\x01\x01 \x00\t$\x01\x00\x02\x08U\x00\x00\x08$\n\x01\x03\x03\x00\x00\x11$\x02\x03\x01\x01\x00\x01\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x11$\x02\x05\x01\x02\x00\x01\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x0c$\x03\x04\x01\x03\x00\x02\x01\x00\x00\x00\x0c$\x03\x06\x01\x01\x00\x05\x01\x00\x00\x00\t\x04\x01\x00\x00\x01\x02 \x00\t\x04\x01\x01\x02\x01\x02 \x00\x10$\x01\x03\x00\x01\x01\x00\x00\x00\x02\x03\x00\x00\x00\x00\x06$\x02\x01\x03\x18\x07\x05\x03\x05\x96\x00\x01\x08%\x01\x00\x00\x00\x00\x00\x07\x05\x83\x11\x04\x00\x04\t\x04\x01\x02\x02\x01\x02 \x00\x10$\x01\x03\x00\x01\x01\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x06$\x02\x01\x03\x18\x07\x05\x03\x05\x84\x03\x01\x08%\x01\x00\x00\x00\x00\x00\x07\x05\x83\x11\x04\x00\x04\t\x04\x02\x00\x00\x01\x02 \x00\t\x04\x02\x01\x01\x01\x02 \x00\x10$\x01\x06\x00\x01\x01\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x06$\x02\x01\x03\x18\x07\x05\x84\x05\x84\x03\x01\x08%\x01\x00\x00\x00\x00\x00\t\x04\x02\x02\x01\x01\x02 \x00\x10$\x01\x06\x00\x01\x01\x00\x00\x00\x02\x03\x00\x00\x00\x00\x06$\x02\x01\x03\x18\x07\x05\x84\x05\x96\x00\x01\x08%\x01\x00\x00\x00\x00\x00\t\x04\x03\x00\x02\x01\x03\x00\x02\x07$\x01\x00\x01a\x00\t$\x03\x01\x03\x01\x02\x01\x04\x06$\x02\x02\x02\x04\t$\x03\x01\x07\x01\x06\x01\x05\x06$\x02\x02\x06\x05\x06$\x02\x01\x01\x04\t$\x03\x02\x04\x01\x01\x01\x04\x06$\x02\x01\x05\x05\t$\x03\x02\x08\x01\x05\x01\x05\t\x05\x07\x02\x00\x02\x00\x00\x00\x06%\x01\x02\x01\x05\t\x05\x86\x02\x00\x02\x00\x00\x00\x06%\x01\x02\x03\x07'
-- Storing configuration <USBConfiguration index=1 num_interfaces=4 attributes=0x80 max_power=100mA> --
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x00), index=0, length=255)
[14:19:06] <: ̄Љ
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=255)
[14:19:06] <: ̰Babyface Pro (72993879)
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x01), index=409, length=255)
[14:19:06] <: ̈RME
[14:19:06] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
[14:19:06] <: ̠0950E421106A7C8
[14:19:06] >, standard request to device (SET_CONFIGURATION: value=1, index=0, length=0)
-- Applying configuration <USBConfiguration index=1 num_interfaces=4 attributes=0x80 max_power=100mA> --
Traceback (most recent call last):
  File "./facedancer-usbproxy.py", line 50, in <module>
    main()
  File "./facedancer-usbproxy.py", line 44, in main
    d.run()
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/USBDevice.py", line 165, in run
    self.scheduler.run()
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/core.py", line 509, in run
    task()
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/USBDevice.py", line 83, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/backends/greatdancer.py", line 765, in service_irqs
    self._handle_setup_events()
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/backends/greatdancer.py", line 368, in _handle_setup_events
    self._handle_setup_event_on_endpoint(i)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/backends/greatdancer.py", line 402, in _handle_setup_event_on_endpoint
    self.connected_device.handle_request(request)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/USBProxy.py", line 241, in handle_request
    self._proxy_out_request(req)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/USBProxy.py", line 296, in _proxy_out_request
    req, data = f.filter_control_out(req, data)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/filters/standard.py", line 95, in filter_control_out
    self.device.configured(configuration)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/USBProxy.py", line 220, in configured
    self.maxusb_app.configured(configuration)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/backends/greatdancer.py", line 740, in configured
    self._configure_endpoints(configuration)
  File "/home/bollie/PycharmProjects/Facedancer/facedancer/backends/greatdancer.py", line 730, in _configure_endpoints
    self.api.set_up_endpoints(*endpoint_triplets)
  File "/home/bollie/PycharmProjects/Facedancer/venv/lib/python3.8/site-packages/pygreat/comms.py", line 1108, in method
    return self.execute_command(verb_number, in_format, out_format, name=name, class_name=class_name,
  File "/home/bollie/PycharmProjects/Facedancer/venv/lib/python3.8/site-packages/pygreat/comms.py", line 1263, in execute_command
    return self.comms_backend.execute_command(self.CLASS_NUMBER, verb, in_format,
  File "/home/bollie/PycharmProjects/Facedancer/venv/lib/python3.8/site-packages/pygreat/comms.py", line 770, in execute_command
    raw_result = self.execute_raw_command(class_number, verb, payload, timeout,
  File "/home/bollie/PycharmProjects/Facedancer/venv/lib/python3.8/site-packages/pygreat/comms_backends/usb.py", line 391, in execute_raw_command
    future_utils.raise_from(self._exception_for_command_failure(error_number, pretty_name), None)
  File "/home/bollie/PycharmProjects/Facedancer/venv/lib/python3.8/site-packages/future/utils/__init__.py", line 403, in raise_from
    exec(execstr, myglobals, mylocals)
  File "<string>", line 1, in <module>
pygreat.comms.CommandFailureError: greatdancer.set_up_endpoints: greatdancer: failing to set up endpoint with packet size 900; larger than max (512)! [EINVAL]
`

FreeDancer is checked out from master, GreatFET is 2020.1.2

Any ideas, why this is failing?

MrBollie, 2020-03-26:
Am I barking up the wrong tree here? This seem to come from https://github.com/greatscottgadgets/greatfet/blob/fbbf4ad5e1abe582fd07bb51078d18dd5d9dfc57/firmware/greatfet_usb/classes/greatdancer.c#L226

And I assume, we're dealing with an isochronous endpoint here and my question is now, will there be support proxying those? In my current scnario, I'm not interested in any audio samples from that device. I'd rather like to see if there are any bulk transfers controlling the DSP hardware mixer build into the audio interface.

dgh05t, 2020-12-02:
same problem here, for proxying a Mass Storage, the Host USB device is a MAC.
log:
```
Using GreatDancer backend.
GreatDancer initialized
[21:59:29] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[21:59:29] <: b'\x12\x01\x10\x03\x00\x00\x00\tQ\tf\x16\x10\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[21:59:30] <: b'\x12\x01\x10\x03\x00\x00\x00\t'
-- Patched device descriptor. --
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[21:59:30] <: b'\x12\x01\x10\x03\x00\x00\x00\tQ\tf\x16\x10\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=2)
[21:59:30] <: ̢
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=34)
[21:59:30] <: ̢DataTraveler 3.0
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x01), index=409, length=2)
[21:59:30] <: ̒
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x01), index=409, length=18)
[21:59:30] <: ̒Kingston
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=2)
[21:59:30] <: ̲
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=50)
[21:59:30] <: ̲20CF302E2382E381163CE238
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=unknown descriptor 0xf00 descriptor (index=0x00), index=0, length=5)
[21:59:30] <: b'\x05\x0f\x16\x00\x02'
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=unknown descriptor 0xf00 descriptor (index=0x00), index=0, length=22)
[21:59:30] <: b'\x05\x0f\x16\x00\x02\x07\x10\x02\x06\x00\x00\x00\n\x10\x03\x00\x0e\x00\x02\n\xff\x07'
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
[21:59:30] <: b'\t\x02,\x00\x01\x01\x00\x80?'
[21:59:30] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=44)
[21:59:30] <: b'\t\x02,\x00\x01\x01\x00\x80?\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x04\x00\x060\x03\x00\x00\x00\x07\x05\x02\x02\x00\x04\x00\x060\x03\x00\x00\x00'
-- Storing configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=126mA> --
[21:59:30] >, standard request to device (SET_CONFIGURATION: value=1, index=0, length=0)
-- Applying configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=126mA> --
Traceback (most recent call last):
  File "facedancer-usbproxy.py", line 50, in <module>
    main()
  File "facedancer-usbproxy.py", line 44, in main
    d.run()
  File "/home/dgh05t/tools/Facedancer-master/facedancer/USBDevice.py", line 165, in run
    self.scheduler.run()
  File "/home/dgh05t/tools/Facedancer-master/facedancer/core.py", line 509, in run
    task()
  File "/home/dgh05t/tools/Facedancer-master/facedancer/USBDevice.py", line 83, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/dgh05t/tools/Facedancer-master/facedancer/backends/greatdancer.py", line 765, in service_irqs
    self._handle_setup_events()
  File "/home/dgh05t/tools/Facedancer-master/facedancer/backends/greatdancer.py", line 368, in _handle_setup_events
    self._handle_setup_event_on_endpoint(i)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/backends/greatdancer.py", line 402, in _handle_setup_event_on_endpoint
    self.connected_device.handle_request(request)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/USBProxy.py", line 241, in handle_request
    self._proxy_out_request(req)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/USBProxy.py", line 296, in _proxy_out_request
    req, data = f.filter_control_out(req, data)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/filters/standard.py", line 95, in filter_control_out
    self.device.configured(configuration)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/USBProxy.py", line 220, in configured
    self.maxusb_app.configured(configuration)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/backends/greatdancer.py", line 740, in configured
    self._configure_endpoints(configuration)
  File "/home/dgh05t/tools/Facedancer-master/facedancer/backends/greatdancer.py", line 730, in _configure_endpoints
    self.api.set_up_endpoints(*endpoint_triplets)
  File "/home/dgh05t/.local/lib/python3.8/site-packages/pygreat/comms.py", line 1108, in method
    return self.execute_command(verb_number, in_format, out_format, name=name, class_name=class_name,
  File "/home/dgh05t/.local/lib/python3.8/site-packages/pygreat/comms.py", line 1263, in execute_command
    return self.comms_backend.execute_command(self.CLASS_NUMBER, verb, in_format,
[... 18 lines cut ...]
[21:58:10] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[21:58:10] <: b'\x12\x01\x10\x03\x00\x00\x00\tQ\tf\x16\x10\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
[21:58:10] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=255)
[... cut ...]
[21:58:12] <: b'\x12\x01\x10\x03\x00\x00\x00\tQ\tf\x16\x10\x01\x01\x02\x03\x01'
-- Patched device descriptor. --
[21:58:12] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=255)
[21:58:12] <: b'\t\x02,\x00\x01\x01\x00\x80?\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x04\x00\x060\x03\x00\x00\x00\x07\x05\x02\x02\x00\x04\x00\x060\x03\x00\x00\x00'
-- Storing configuration <USBConfiguration index=1 num_interfaces=1 attributes=0x80 max_power=126mA> --
```

straithe, 2021-11-13:
Hi, @MrBollie. Are you still experiencing this issue?

closed 2021-12-12 by straithe