greatscottgadgets/facedancer #47: Trying to facedancer-usbproxy.py a Roland Boutique D-05 sound module

jaffa225man, 2020-08-06:
It's probably silly, but I just purchased the GreatFET One to try illuminating what the ALSA driver is missing, that windows isn't, to get the Roland Boutique D-05's USB digital audio working under GNU/Linux.

Since facedancer-usbproxy.py didn't work at all with the 2018 firmware the GreatFET One arrived with, I updated its firmware to the current git:

```
gf info
Found a GreatFET One!
  Board ID: 0
  Firmware version: git-v2020.1.2-18-g12efb0f
  Part ID: a0000a30724f6e
  Serial number: 000057cc67e631177857
```

With the current firmware I definitely have gotten a lot farther, as it waits for me to plug into the target (windows computer) before crashing.  A USB keyboard, rather than the D-05, works perfectly, by the way.  

This is what the Roland Boutique D-05 looks like to lsusb -v:
```
Bus 001 Device 005: ID 0582:01ff Roland Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol       255 
  bMaxPacketSize0        64
  idVendor           0x0582 Roland Corp.
  idProduct          0x01ff 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x010e
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      2 
      bInterfaceProtocol      2 
      iInterface              0 
      ** UNRECOGNIZED:  06 24 f1 01 00 00
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
[... 187 lines cut ...]
    self.connected_device.handle_request(request)
  File "/usr/src/Facedancer/facedancer/USBProxy.py", line 239, in handle_request
    self._proxy_in_request(req)
  File "/usr/src/Facedancer/facedancer/USBProxy.py", line 277, in _proxy_in_request
    req, data, stalled = f.filter_control_in(req, data, stalled)
  File "/usr/src/Facedancer/facedancer/filters/standard.py", line 51, in filter_control_in
    configuration = USBDescribable.from_binary_descriptor(data)
  File "/usr/src/Facedancer/facedancer/USB.py", line 83, in from_binary_descriptor
    return subclass.from_binary_descriptor(data)
  File "/usr/src/Facedancer/facedancer/USBConfiguration.py", line 58, in from_binary_descriptor
    interfaces = cls._parse_subordinate_descriptors(data[length:total_length])
  File "/usr/src/Facedancer/facedancer/USBConfiguration.py", line 78, in _parse_subordinate_descriptors
    descriptor = USBDescribable.from_binary_descriptor(data[:length])
  File "/usr/src/Facedancer/facedancer/USB.py", line 83, in from_binary_descriptor
    return subclass.from_binary_descriptor(data)
  File "/usr/src/Facedancer/facedancer/USBInterface.py", line 82, in from_binary_descriptor
    = struct.unpack_from("xxBBBBBBB", data)
struct.error: unpack_from requires a buffer of at least 9 bytes for unpacking 9 bytes at offset 0 (actual buffer size is 8)
```

I've tried editing facedancer here and there, but so far haven't found anything that keeps the device descriptor data and works.  Aside from looking at python in the past, this is my first real attempt editing it, though.  This problem looked different enough from what others have posted, so I hope there's an easy, straightforward solution.

Thanks for any ideas you can offer,

  Luke

jaffa225man, 2021-07-11:
Although the issue shown here probably still exists, I don't need it solved (for this purpose/device anyway) because many Roland devices are now working for USB-audio on recent kernels with auto-detection and configuration.  That's due to the amazing dedication of, ALSA kernel developer, Takashi Iwai!: https://patchwork.kernel.org/project/alsa-devel/patch/20210422120413.457-2-tiwai@suse.de/

straithe, 2021-11-13:
Thank you for the update. @jaffa225man and I'm sorry we didn't get to this issue while it was still fresh.

closed 2021-11-13 by straithe