greatscottgadgets/facedancer #49: Help setting up this USB device for emulation

Eddiiie, 2020-09-07:
Hello,

I bought a GreatFet and a total phase Beagle 480 a year or so ago to help try to emulate this usb device used on a closed system.
The USB protocol is new to me.  I've read through 100 pages of the USB spec.. The information the Total Phase Data Center captures is nice but it I cannot really understand what I am looking at yet.  From this screenshot, can someone extract the required information needed for FaceDancer and the template to get something communicating?  

Attached is a screenshot of where I think the device descriptor and device configuration is displayed
[image]
[image]

[Capture1_Data.zip](https://github.com/usb-tools/Facedancer/files/5180944/Capture1_Data.zip)

ktemkin, 2020-09-07:
I unfortunately don't have the bandwidth to go through those datasets, but I can point you towards some resources. When looking at the USB spec, you'll mostly want to look through Chapter 9 (the USB device framework). 

More expansively, I have a bunch of resources on the webpage for my USB training that might help:  https://usbc.tf/

straithe, 2021-11-13:
Hi, @Eddiiie. Do you still need assistance?

Eddiiie, 2021-11-17:
@straithe  Wowee  sure do!  it has been so long since I opened this I'd have to shuffle things around and go back, but happy to do it.  Wound up taking another approach and went way down the rabbit hole.      Since then I must have read the usb spec (the important pages) at least 5 times and I think it is starting to finally sink in.   Transactions, End Points, still pretty foggy though.  Appreciate any help. If we can exchange some emails?     Thank you.

straithe, 2021-11-29:
GitHub works best for me as it is easier for me to get multiple people on my team to help out if needed.

To make sure I understand your issue correctly I'm going to repeat back:

What you want is help figuring out how to get a certain device communicating. Is that correct? Can you give me more detail?

closed 2021-12-28 by straithe