greatscottgadgets/facedancer #56: error from initialize_device in

pk-r7, 2021-03-16:
I am trying to proxy a device using a GF one. 
I validated this on 2 separate operating systems, Kali and Ubuntu, virtual and bare metal, respectively. The Ubuntu installation is using Python 3.6.9 and Kali is using Python 3.8.2
I have validated I can see the device using the sample code from the Greatfet issues, so the hardware appears to be working.
`import greatfet

gf = greatfet.GreatFET()
descriptor = gf.glitchkit.usb.capture_control_in(
             request= gf.glitchkit.usb.GET_DESCRIPTOR,
             value  = gf.glitchkit.usb.GET_DEVICE_DESCRIPTOR,
             length = 18
         )
print(descriptor)
b'\x12\x01\x00\x02\x00\x00\x00@P\x1d\x89`\x03\x01\x01\x02\x04\x01'
`
However using the function in Facedancer generates the error:
`
u = FacedancerUSBHostApp(verbose=3)
Using GreatDancer Host backend.
>>> u.initialize_device(assign_address=1, apply_configuration=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/facedancer-2.9-py3.8.egg/facedancer/core.py", line 365, in initialize_device
  File "/usr/local/lib/python3.8/dist-packages/facedancer-2.9-py3.8.egg/facedancer/core.py", line 401, in read_ep0_max_packet_size
  File "/usr/local/lib/python3.8/dist-packages/facedancer-2.9-py3.8.egg/facedancer/core.py", line 395, in get_device_descriptor
  File "/usr/local/lib/python3.8/dist-packages/facedancer-2.9-py3.8.egg/facedancer/core.py", line 386, in get_descriptor
  File "/usr/local/lib/python3.8/dist-packages/facedancer-2.9-py3.8.egg/facedancer/core.py", line 291, in control_request_in
  File "/usr/local/lib/python3.8/dist-packages/facedancer-2.9-py3.8.egg/facedancer/backends/greathost.py", line 369, in send_on_endpoint
OSError: Stalled!
`

Qyriad, 2021-03-16:
If you're trying to *proxy* a device, then you should be using `FacedancerUSBApp` and `USBProxyDevice` instead of `FacedancerUSBHostApp`. Look at scripts like [`facedancer-usbproxy.py`](https://github.com/usb-tools/Facedancer/blob/master/facedancer-usbproxy.py) and [`usbproxy-switch-invertx.py`](https://github.com/usb-tools/Facedancer/blob/master/usbproxy-switch-invertx.py) for examples.

pk-r7, 2021-03-16:
True, but I was just trying to enumerate the device so I could confirm USB1 worked, those two scripts I also had issues with and I wanted to confirm I had physical connectivity working before proxying. The OSError seemed to be universal across recent python versions so I wanted to raise the issue.

straithe, 2021-11-13:
Do you still need assistance, @pk-r7 ?

pk-r7, 2021-11-14:
Yes the issue was never resolved.

straithe, 2021-11-14:
Thank you for the update! I don't quite have the skills to assist you, but I will ensure someone on my team takes a look at this soon.

antoinevg, 2025-05-15:
At the risk of doing permanent damage to the definition of the word "soon"…

Does the issue still occur on the `0.3.x` release of Facedancer?

Tx!