greatscottgadgets/facedancer #58: cant proxy PSVita? NotImplementedError: detach_kernel_driver

KuromeSan, 2021-09-04:
C:\Users\Silica\Desktop\greatfet\Facedancer>py facedancer-usbproxy.py -v 054C -p 04E4
Using GreatDancer backend.
GreatDancer initialized
Traceback (most recent call last):
  File "C:\Users\Silica\Desktop\greatfet\Facedancer\facedancer-usbproxy.py", line 50, in <module>
    main()
  File "C:\Users\Silica\Desktop\greatfet\Facedancer\facedancer-usbproxy.py", line 33, in main
    d = USBProxyDevice(u, idVendor=args.vendorid, idProduct=args.productid, verbose=2, quirks=quirks)
  File "C:\Users\Silica\Desktop\greatfet\Facedancer\facedancer\USBProxy.py", line 173, in __init__
    self.libusb_device.detach_kernel_driver(interface.bInterfaceNumber)
  File "C:\Users\Silica\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 1121, in detach_kernel_driver
    self._ctx.backend.detach_kernel_driver(
  File "C:\Users\Silica\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\backend\libusb0.py", line 662, in detach_kernel_driver
    raise NotImplementedError(self.detach_kernel_driver.__name__)
NotImplementedError: detach_kernel_driver

KuromeSan, 2021-09-06:
tried it on linux, and got further, but now my ps3 just stops responding to any usb stuff, 
not even just the psvita, 

also didnt seem to work with the DualShock 3 ..

straithe, 2021-11-13:
Are you still experiencing this issue, @KuromeSan ?

KuromeSan, 2021-11-13:
Yes

straithe, 2021-11-13:
Thank you for the update! I am quite certain how to assist you myself, but I will make sure my team takes a look at this when they can.