greatscottgadgets/facedancer #61: Proxying usb gamecontroller with EP 0x04 / EP 0x84: failed to configure impossible endpoint with address 4 [EINVAL]

dominikfehr, 2022-03-02:
BACKEND: `greatfet`
Facedancer commit: e688fe61dc34087db333432394e1f90e52ac3794
GreatFET firmware: `Firmware version: v2021.2.1`

## Problem

Trying to proxy a XBox One Game-controller via FaceDancer, error received is:

`greatdancer.set_up_endpoints: greatdancer: failed to configure impossible endpoint with address 4 [EINVAL]`

Also tried older Facedancer repo states (end 2019)...

Any hints?

### Error output
```
$ sudo -E python3 facedancer-usbproxy.py -v 045e -p 0b12
Using GreatDancer backend.
GreatDancer initialized
[20:55:50] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=64)
[20:55:50] <: b'\x12\x01\x00\x02\xffG\xd0@^\x04\x12\x0b\x07\x05\x01\x02\x03\x01'
-- Patched device descriptor. --
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[20:55:51] <: b'\x12\x01\x00\x02\xffG\xd0@^\x04\x12\x0b\x07\x05\x01\x02\x03\x01'
-- Patched device descriptor. --
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=255)
[20:55:51] <: b'\t\x02w\x00\x03\x01\x00\xa0\xfa\t\x04\x00\x00\x02\xffG\xd0\x00\x07\x05\x02\x03@\x00\x04\x07\x05\x82\x03@\x00\x04\t\x04\x00\x01\x02\xffG\xd0\x00\x07\x05\x02\x03@\x00\x04\x07\x05\x82\x03@\x00\x02\t\x04\x01\x00\x00\xffG\xd0\x00\t\x04\x01\x01\x02\xffG\xd0\x00\x07\x05\x03\x01\xe4\x00\x01\x07\x05\x83\x01@\x00\x01\t\x04\x02\x00\x00\xffG\xd0\x00\t\x04\x02\x01\x02\xffG\xd0\x00\x07\x05\x04\x02@\x00\x00\x07\x05\x84\x02@\x00\x00'
-- Storing configuration <USBConfiguration index=1 num_interfaces=3 attributes=0xA0 max_power=500mA> --
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
[20:55:51] <: ̺30393731323236373333423524525
[20:55:51] <, vendor request to device (vendor request 144: value=0, index=4, length=16)
[20:55:51] <: b'(\x00\x00\x00\x00\x01\x04\x00\x01\x00\x00\x00\x00\x00\x00\x00'
[20:55:51] <, vendor request to device (vendor request 144: value=0, index=4, length=40)
[20:55:51] <: b'(\x00\x00\x00\x00\x01\x04\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x01XGIP10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x00), index=0, length=255)
[20:55:51] <: ̄Љ
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=255)
[20:55:51] <: ̖Controller
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=DEVICE_QUALIFIER descriptor (index=0x00), index=0, length=10)
[20:55:51] < --STALLED-- 
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[20:55:51] <: b'\x12\x01\x00\x02\xffG\xd0@^\x04\x12\x0b\x07\x05\x01\x02\x03\x01'
-- Patched device descriptor. --
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
[20:55:51] <: b'\t\x02w\x00\x03\x01\x00\xa0\xfa'
[20:55:51] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=119)
[20:55:51] <: b'\t\x02w\x00\x03\x01\x00\xa0\xfa\t\x04\x00\x00\x02\xffG\xd0\x00\x07\x05\x02\x03@\x00\x04\x07\x05\x82\x03@\x00\x04\t\x04\x00\x01\x02\xffG\xd0\x00\x07\x05\x02\x03@\x00\x04\x07\x05\x82\x03@\x00\x02\t\x04\x01\x00\x00\xffG\xd0\x00\t\x04\x01\x01\x02\xffG\xd0\x00\x07\x05\x03\x01\xe4\x00\x01\x07\x05\x83\x01@\x00\x01\t\x04\x02\x00\x00\xffG\xd0\x00\t\x04\x02\x01\x02\xffG\xd0\x00\x07\x05\x04\x02@\x00\x00\x07\x05\x84\x02@\x00\x00'
-- Storing configuration <USBConfiguration index=1 num_interfaces=3 attributes=0xA0 max_power=500mA> --
[20:55:51] <, standard request to device (GET_STATUS: value=0, index=0, length=2)
[20:55:51] <: b'\x02\x00'
[20:55:51] >, standard request to device (SET_CONFIGURATION: value=1, index=0, length=0)
-- Applying configuration <USBConfiguration index=1 num_interfaces=3 attributes=0xA0 max_power=500mA> --
Traceback (most recent call last):
  File "/home/userA/Desktop/Facedancer/facedancer-usbproxy.py", line 50, in <module>
    main()
  File "/home/userA/Desktop/Facedancer/facedancer-usbproxy.py", line 44, in main
    d.run()
  File "/home/userA/Desktop/Facedancer/facedancer/USBDevice.py", line 165, in run
    self.scheduler.run()
  File "/home/userA/Desktop/Facedancer/facedancer/core.py", line 509, in run
    task()
  File "/home/userA/Desktop/Facedancer/facedancer/USBDevice.py", line 83, in <lambda>
    self.scheduler.add_task(lambda : self.maxusb_app.service_irqs())
  File "/home/userA/Desktop/Facedancer/facedancer/backends/greatdancer.py", line 765, in service_irqs
    self._handle_setup_events()
  File "/home/userA/Desktop/Facedancer/facedancer/backends/greatdancer.py", line 368, in _handle_setup_events
    self._handle_setup_event_on_endpoint(i)
  File "/home/userA/Desktop/Facedancer/facedancer/backends/greatdancer.py", line 402, in _handle_setup_event_on_endpoint
    self.connected_device.handle_request(request)
  File "/home/userA/Desktop/Facedancer/facedancer/USBProxy.py", line 241, in handle_request
[... 178 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

tuxuser, 2022-03-04:
In a similar situation, while trying to sniff/analyze a very similar target device.

Target is an original `054c:09cc Sony Corp. DualShock 4 [CUH-ZCT2x]`.

It does, just like the device mentioned above, feature an "impossible endpoint" on interface 5:

```
---snip---
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
--- snip ---
```
## Desired outcome

PS4 Console detects "mitm'd" PS4 Controller and lets you use it.

## Problem situation

Both approaches mentioned below do not yield usable results.

Approach 1) endpoint rerouting, ends in a loop of "request descriptor".
Approach 2) modifying firmware, just stalls the standard `facedancer-usbproxy.py` script.

## Ideas tried so far

### Rewriting the endpoint `0x84` to `0x83` when the target device requests it.

Code used (based on the Nintendo Switch Pro controller rewrite)
```py
#!/usr/bin/env python3
#
# facedancer-switch-flip-endpoint.py

from facedancer import FacedancerUSBApp
from facedancer.USBConfiguration import USBConfiguration
from facedancer.USBInterface import USBInterface
from facedancer.USBEndpoint import USBEndpoint
from facedancer.USBProxy import USBProxyDevice, USBProxyFilter
from facedancer.filters.standard import USBProxySetupFilters
from facedancer.filters.logging import USBProxyPrettyPrintFilter

from enum import Enum

GET_DESCRIPTOR_REQUEST = 6
DESCRIPTOR_CONFIGURATION = 0x02

TARGET_INTERFACE = 5

EP4_IN_ORIG = 0x04 # 0x84
EP4_IN_MOD  = 0x03 # 0x83

USB_VID, USB_PID = 0x054c, 0x09cc
[... 778 lines cut ...]
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Status:     0x0000
  (Bus Powered)
```

jaysonlarose, 2023-03-29:
Did you ever find a good workaround for this? Also, what is this "Nintendo Switch Pro controller rewrite"?  I'm asking because I'm looking to reliably capture controller data sent between video game controllers and the console... specifically the Nintendo Switch for now, but ideally I'd like a solution that works on any USB controller...

dominikfehr, 2023-03-29:
> Did you ever find a good workaround for this? Also, what is this "Nintendo Switch Pro controller rewrite"? I'm asking because I'm looking to reliably capture controller data sent between video game [...]

hey @jaysonlarose, at one point we stopped using the greatfet hardware due to too many problems. for pure USB sniffing I can recommend http://openvizsla.org/

there's also another gsg hardware upcoming which might be more sufficent than greatfet: https://greatscottgadgets.com/cynthion/

jaysonlarose, 2023-03-29:
Hi @dominikfehr !
Interesting. Is it capable of outputting a realtime dump of the USB data going across the line?

mossmann, 2025-09-18:
It looks like Cynthion (which is now released) would be suitable for this project. It supports arbitrary endpoint configuration and also can be used as a USB analyzer.

https://greatscottgadgets.com/cynthion/