greatscottgadgets/facedancer #62: HID Mouse stuck at 125 Hz + Crashing @ 1000 Hz

gionetti, 2022-05-30:
I wrote a simple HID mouse class for Facedancer. The basic functionality works as expected. 

**Issue:**
Regardless of the devices `bInterval` setting, the device seems to be stuck at a max of 125 Hz (60 Hz on average) 
Tested using https://devicetests.com/mouse-rate-test

Any interval below 4ms will cause an error shortly after EP1 is configured. 

What is my best strategy for debugging this? Is my device configuration causing the issue, such as a missing IN or OUT handler?

**Here is my HID class and the stack trace for reference:**
```
from .                        import default_main

from ..future                 import *
from ..classes.hid.usage      import *
from ..classes.hid.descriptor import *
from ..classes.hid.mouse      import *
from ..classes                import USBDeviceClass

@use_inner_classes_automatically
class USBMouseDevice(USBDevice):
    """ Simple USB mouse device. """
    name                     : str   = "Wired Mouse"
    device_class             : int   = 0
    device_subclass          : int   = 0
    protocol_revision_number : int   = 0
    max_packet_size_ep0      : int   = 64
    vendor_id                : int   = 0x1234 
    product_id               : int   = 0x5678
    manufacturer_string      : str   = "Company, Inc."
    product_string           : str   = "ABC Mouse"
    serial_number_string     : str   = "12345678"
    supported_languages      : tuple = (LanguageIDs.ENGLISH_US,)
    device_revision          : int   = 0
    usb_spec_version         : int   = 0x0002

    class MouseConfiguration(USBConfiguration):
        """ Primary USB configuration: act as a mouse. """
        number                 : int    = 1
        configuration_string   : str    = None
        max_power              : int    = 500
        self_powered           : bool   = True
        supports_remote_wakeup : bool   = True

        class MouseInterface(USBInterface):
            """ Core HID interface for our mouse. """
            name                   : str = "USB mouse interface"
            number                 : int = 0
            alternate              : int = 0
            class_number           : int = USBDeviceClass.HID
            subclass_number        : int = 0
            protocol_number        : int = 0
            interface_string       : str = None

            class MouseEventEndpoint(USBEndpoint):
                number               : int                    = 1
                direction            : USBDirection           = USBDirection.IN
                transfer_type        : USBTransferType        = USBTransferType.INTERRUPT
                synchronization_type : USBSynchronizationType = USBSynchronizationType.NONE
                usage_type           : USBUsageType           = USBUsageType.DATA
                max_packet_size      : int                    = 64
                interval             : int                    = 4

            class USBClassDescriptor(USBClassDescriptor):
                number      : int   =  USBDescriptorTypeNumber.HID
                raw         : bytes = b'\x09\x21\x10\x01\x00\x01\x22\x3A\x00'
                # bLength = 0x09 = 9 bytes
                # bDescriptorType = 0x21 = 33 = HID
                # bcdHID = 0x0110 = 1.1 HID spec release version (little endian)
[... 139 lines cut ...]
    self.handle_data_requested(endpoint)
  File "D:\Anaconda3\lib\site-packages\facedancer\devices\mouse.py", line 167, in handle_data_requested
    endpoint.send(report)
  File "D:\Anaconda3\lib\site-packages\facedancer\future\endpoint.py", line 78, in send
    packet_size=self.max_packet_size, blocking=blocking)
  File "D:\Anaconda3\lib\site-packages\facedancer\future\device.py", line 251, in _send_in_packets
    self.backend.send_on_endpoint(endpoint_number, packet, blocking=blocking)
  File "D:\Anaconda3\lib\site-packages\facedancer\backends\greatdancer.py", line 246, in send_on_endpoint
    self.api.send_on_endpoint(ep_num, bytes(data))
  File "D:\Anaconda3\lib\site-packages\pygreat\comms.py", line 1109, in method
    timeout=timeout, max_response_length=max_response_length, encoding=encoding, *arguments)
  File "D:\Anaconda3\lib\site-packages\pygreat\comms.py", line 1264, in execute_command
    out_format, *arguments, **kwargs)
  File "D:\Anaconda3\lib\site-packages\pygreat\comms.py", line 771, in execute_command
    None, max_response_length, comms_timeout, pretty_name, rephrase_errors)
  File "D:\Anaconda3\lib\site-packages\pygreat\comms_backends\usb.py", line 353, in execute_raw_command
    self.LIBGREAT_REQUEST_NUMBER, self.LIBGREAT_VALUE_EXECUTE, flags, to_send, timeout)
  File "D:\Anaconda3\lib\site-packages\usb\core.py", line 1089, in ctrl_transfer
    self.__get_timeout(timeout))
  File "D:\Anaconda3\lib\site-packages\usb\backend\libusb0.py", line 609, in ctrl_transfer
    timeout
  File "D:\Anaconda3\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: A device attached to the system is not functioning.\r\n\n'
```

martinling, 2022-05-30:
This doesn't seem surprising to me.

Each time the target host polls your mouse for a report, the firmware running on the GreatFET has to send a request back to the control host. It can't do that instantly - it has to wait for the control host to poll it for data. The control host has to see that request and execute your `handle_data_requested` method. Then the result gets sent back to the GreatFET, which can then respond to the the target host. This round trip happens every time the target host polls your mouse for a report, and takes a few milliseconds.

The GreatFET itself should be able to handle sending reports at 1000Hz, but only if it has them ready to send. To hit that rate, you'd need to arrange things such that the control host is sending its reports to the GreatFET in advance, and the GreatFET is sending reports one at a time from a queue. I expect that would require some changes to Facedancer.

gionetti, 2022-06-08:
First off, @martinling thank you for taking time to explain to me the reason for the observed frequency. This really helped me understand where my problem was. Also, cloning the firmware repo, facedancer repo, and reading the basics on https://www.beyondlogic.org/usbnutshell/usb1.shtml was really crucial for me to understand how this emulation is happening and how USB works in general.

I have partially solved my problem without touching the firmware of the device. 

1) The `handle_data_requested` has been replaced with the 'handle_buffer_empty'. This increased the frequency that device could send packets to target host by a factor of 2 roughly as we are no longer waiting for the NAK interrupt that the device raises. We operate when the device has an empty buffer ready to accept new DATA for target host to read. (Host polls are not missing)

2) the `endpoint.send()` method has been replaced by a custom method that uses assumptions about how this HID device functions to improve directness and speed

3) the IRQ routine for greatdancer.py has been modified to prioritize EP1 IN (specific to this use case)

4) the SetPriorityClass and SetThreadPriority methods from win32 have been used to request that the process and threads are given higher scheduling priority from an OS perspective

Overall I have achieved a frequency of 500Hz and the device no longer crashes (due to timeout) when host is polling at 1ms interval. To go beyond this, modifications to the usb_queue and IRQ handlers in the firmware would need to be made as you mentioned in your comment. If the DATA for the interrupt endpoint is already on the device and ready to send to host frequencies of 1000Hz+ could be achieved.