greatscottgadgets/facedancer #64: Problem running USBKeyboard.py using the GreatFet One on Windows 10

C01J, 2022-06-24:
Hi,

I am using the GreatFet One on Windows 10 with the Firmware version: v2021.2.1.

I have been trying to run the facedancer's script `[USBKeyboard.py]` from `(https://github.com/greatscottgadgets/Facedancer/blob/master/legacy-applets/USBKeyboard.py)`.

However, I have been encountering the following problem when I connect USB1:

[image]
[image]

The printout are as follow:
[image]

Kindly advise on how to resolve it. Thank you.

nitzanfarhi, 2023-03-09:
Happens to me as well

antoinevg, 2024-08-15:
We are in the process of adding Windows support to GreatFET, will keep y'all updated!