greatscottgadgets/facedancer #82: Stall when Proxying USB Packets with iPhone

b3ll, 2024-01-25:
I'm trying to proxy and read the HID packets from an iPhone 11 Pro (to a car stereo) when the iPhone has its configuration changed to the HID one, however it just ends up stalling and doesn't progress any further.

I'm using the following to start: `sudo ./facedancer-usbproxy.py -v 05ac -p 12a8` and it does change the configuration to 2 (which is the HID one), and then it seems to request the descriptor for HID (which is in fact 208 bytes)

I'm using a raspberry pi 4b (on the latest raspbian) and a GreatFET One to do this. I tried it on macOS as well to the same issue. I'm not sure what I should try next…

My setup is like this:

[iPhone] -> [rpi host] 
[rpi host] -> [GreatFET host] 
[Car Stereo] -> [GreatFET target]

Here's what it prints out:

```
Using GreatDancer backend.
GreatDancer initialized
-- backend does not support setting device speed: HIGH --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=8)
[16:30:42] <: b'\x12\x01\x00\x02\x00\x00\x00@'
-- Patched device descriptor. --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=DEVICE descriptor (index=0x00), index=0, length=18)
[16:30:42] <: b'\x12\x01\x00\x02\x00\x00\x00@\xac\x05\xa8\x12\x05\x12\x01\x02\x03\x04'
-- Patched device descriptor. --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=9)
[16:30:42] <: b"\t\x02'\x00\x01\x01\x05\xc0\xfa"
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x00), index=0, length=39)
[16:30:42] <: b"\t\x02'\x00\x01\x01\x05\xc0\xfa\t\x04\x00\x00\x03\x06\x01\x01\x0f\x07\x05\x02\x02\x00\x02\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x83\x03@\x00\n"
-- Storing configuration <USBConfiguration index=1 num_interfaces=1 attributes=0xC0 max_power=500mA> --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x01), index=0, length=9)
[16:30:42] <: b'\t\x02\x95\x00\x03\x02\x06\xc0\xfa'
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x01), index=0, length=149)
[16:30:42] <: b'\t\x02\x95\x00\x03\x02\x06\xc0\xfa\t\x04\x00\x00\x00\x01\x01\x00\x00\t$\x01\x00\x01\x1e\x00\x01\x01\x0c$\x02\x01\x01\x02\x02\x02\x03\x00\x00\x00\t$\x03\x02\x01\x01\x01\x01\x00\t\x04\x01\x00\x00\x01\x02\x00\x00\t\x04\x01\x01\x01\x01\x02\x00\x00\x07$\x01\x02\x01\x01\x00#$\x02\x01\x02\x02\x10\t@\x1f\x00\x11+\x00\xe0.\x00\x80>\x00"V\x00\xc0]\x00\x00}\x00D\xac\x00\x80\xbb\x00\t\x05\x81\x01\xc0\x00\x04\x00\x00\x07%\x01\x01\x00\x00\x00\t\x04\x02\x00\x01\x03\x00\x00\x00\t!\x11\x01\x00\x01"\xd0\x00\x07\x05\x83\x03@\x00\x01'
-- Storing configuration <USBConfiguration index=2 num_interfaces=3 attributes=0xC0 max_power=500mA> --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x02), index=0, length=9)
[16:30:42] <: b'\t\x02>\x00\x02\x03\x07\xc0\xfa'
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x02), index=0, length=62)
[16:30:42] <: b'\t\x02>\x00\x02\x03\x07\xc0\xfa\t\x04\x00\x00\x03\x06\x01\x01\x0f\x07\x05\x02\x02\x00\x02\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x83\x03@\x00\n\t\x04\x01\x00\x02\xff\xfe\x02\x11\x07\x05\x04\x02\x00\x02\x00\x07\x05\x85\x02\x00\x02\x00'
-- Storing configuration <USBConfiguration index=3 num_interfaces=2 attributes=0xC0 max_power=500mA> --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x03), index=0, length=9)
[16:30:42] <: b'\t\x02u\x00\x03\x04\x08\xc0\xfa'
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=CONFIGURATION descriptor (index=0x03), index=0, length=117)
[16:30:42] <: b'\t\x02u\x00\x03\x04\x08\xc0\xfa\t\x04\x00\x00\x03\x06\x01\x01\x0f\x07\x05\x02\x02\x00\x02\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x83\x03@\x00\n\t\x04\x01\x00\x02\xff\xfe\x02\x11\x07\x05\x04\x02\x00\x02\x00\x07\x05\x85\x02\x00\x02\x00\t\x04\x02\x00\x00\xff\xfd\x01\x19\t\x04\x02\x01\x02\xff\xfd\x01\x19\x07\x05\x86\x02\x00\x02\x00\x07\x05\x05\x02\x00\x02\x00\t\x04\x02\x02\x02\xff\xfd\x01\x19\x07\x05\x86\x02\x00\x02\x00\x07\x05\x05\x02\x00\x02\x00'
-- Storing configuration <USBConfiguration index=4 num_interfaces=3 attributes=0xC0 max_power=500mA> --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=DEVICE_QUALIFIER descriptor (index=0x00), index=0, length=10)
[16:30:42] <: b'\n\x06\x00\x02\x00\x00\x00@\x04\x00'
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x00), index=0, length=255)
[16:30:42] <: ̄Љ
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x02), index=409, length=255)
[16:30:42] <: ̎iPhone
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x03), index=409, length=255)
[16:30:42] <: ͒OMITTED
[16:30:42] >, standard request to device (SET_CONFIGURATION: value=2, index=0, length=0)
-- Applying configuration <USBConfiguration index=2 num_interfaces=3 attributes=0xC0 max_power=500mA> --
[16:30:42] <, standard request to device (GET_DESCRIPTOR: value=STRING descriptor (index=0x05), index=409, length=255)
[16:30:42] <: ̈PTP
[Errno 2] Entity not found
[16:30:42] <, standard request to interface (GET_DESCRIPTOR: value=REPORT descriptor (index=0x00), index=2, length=208)
[16:30:42] < --STALLED--
```

Here's lsusb:

```
Bus 001 Device 005: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE/7/8/X/XR
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
[... 414 lines cut ...]
      bInterfaceProtocol      1
      iInterface             25
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

Any ideas?

antoinevg, 2024-01-25:
Most likely the issue here is that the GreatFET One only supports full-speed USB on the target side but the car stereo is expecting to talk to the iPhone as a high-speed device.

b3ll, 2024-01-25:
Oh snap, I didn’t know that the GreatFET didn’t support high speed :(

It is indeed expecting high speed, thanks for getting back!

Do you have any recs for devices that support high speed? I saw cynthion (and it looks wicked!), but that isn’t shipping till June and I wanted to solve this in a week or two

b3ll, 2024-01-26:
Just following up here, finally got something working (
https://github.com/AristoChen/usb-proxy/tree/main). It isn't pretty, but at least I can read the raw packets for now. Excited to see Cynthion!

closed 2024-01-26 by b3ll

xairy, 2024-01-27:
Yay, Raw Gadget FTW! :)

It doesn't support isochronous endpoints though, so you might get issues with your device. But proxying the HID interface should hopefully work.

b3ll, 2024-01-27:
I actually didn't need isochronous endpoints at all! I just needed the HID interface. It failed the first time, but worked well enough on subsequent attempts