greatscottgadgets/facedancer #108: Is it a bottleneck or just the device capabilites?

MaorRakach, 2024-08-21:
I run the mass-storage.py example and then measured the performance using the linux dd command.
Something like this:
`dd if={mounted_device_path e.g. /dev/sdf} of=/dev/null bs=512 count=100000`

The result was bout 40-60 kb/s on the Cynthion board.

Then I configured the script to run at USB2.0 High speed like this:
```
@use_inner_classes_automatically
class USBMassStorageDevice(USBDevice):
    """ Class implementing an emulated USB Mass Storage device. """

    name                 : str = "USB mass storage interface"

    vendor_id            : int = 0x610b # Sandisk
    product_id           : int = 0x4650 # SDCZ2 Cruzer Mini Flash Drive (thin)
    device_revision      : int = 0x0003

    manufacturer_string  : str = "Facedancer"
    product_string       : str = "USB Mass Storage emulation"

    max_packet_size_ep0  : int = 64
    device_speed         : DeviceSpeed = DeviceSpeed.HIGH

    class _Configuration(USBConfiguration):
        configuration_string : str = "Mass Storage config"

        class _Interface(USBInterface):

            # This is a Mass Storage Class device
            class_number    : int = USBDeviceClass.MASS_STORAGE
            subclass_number : int = 0x06 # SCSI transparent command set
            protocol_number : int = 0x50 # bulk-only (BBB) transport

            class _OutEndpoint(USBEndpoint):
                number          : int             = ENDPOINT_OUT
                direction       : USBDirection    = USBDirection.OUT
                transfer_type   : USBTransferType = USBTransferType.BULK
                max_packet_size : int             = 512

            class _InEndpoint(USBEndpoint):
                number        : int               = ENDPOINT_IN
                direction     : USBDirection      = USBDirection.IN
                transfer_type : USBTransferType   = USBTransferType.BULK
                max_packet_size : int             = 512

    def __init__(self, disk_image):
        self.disk_image = disk_image
        super().__init__()
        
        self.device_speed = DeviceSpeed.HIGH

```

Indeed I achieved a rate of about 140-160 kb/s, 200% increase.

However, I still think that USB 2.0 High Speed can do much better, few MB/s at least.

So, my question do I or the Cynthion/FaceDancer have some bottleneck that I'm now aware of.
Or this definitely close the maximum Cynthion chip capabilites, and there is no much where and how to improve.

Thanks:)

antoinevg, 2024-08-21:
Nicely spotted! We do indeed have a bottleneck with the current implementation of the Cynthion Facedancer SoC<=>LUNA peripheral interface.

While the LUNA USB gateware is quite capable of saturating a 480Mbps High-speed USB bus on Cynthion, the SoC is currently restricted to a register-based interface which only allows the transfer of a single byte per register operation. Given that the SoC is clocked at 60 MHz the constraints of this architecture quickly becomes apparent.

This is an active area of research for us and we're exploring approaches to a memory-based SoC peripheral that can take the load off the CPU and get us closer to the theoretical bandwidth limitations of USB 2.0.

MaorRakach, 2024-08-21:
> Nicely spotted! We do indeed have a bottleneck with the current implementation of the Cynthion Facedancer SoC<=>LUNA peripheral interface.
> 
> While the LUNA USB gateware is quite capable of saturating a 480Mbps High-speed USB link on Cynthion, the SoC is currently restricted to a register-based interface which only allows the transfer of a single byte per register operation. Given that the SoC is clocked at 60 MHz the constraints of this architecture quickly becomes apparent.
> 
> This is an active area of research for us and we're exploring approaches to a memory-based SoC peripheral that can take the load off the CPU and get us closer to the theoretical bandwidth limitations of USB 2.0.

Thanks so much for the explanation, really insightful!

So just to make sure I get it correctly:
1. The FPGA hardware & Luna Gateware logic is enough to handle USB 2.0.
2. The SoC which wraps the FPGA, can not get the bytes out of the FPGA logic "fast" enough.
3. This is something that is unique to the Cynthion board.

Questions:
1. Are there currently any USB2.0 devices that can serve as a face dancer backend and don't have this limitation?
2.  If I understand you correctly, it also means that if the mass storage is all implemented in the FPGA, then that theoretically should solve the problem, however, you still would have to contact external memory (as backend for the mass storage disk) with the SoC, and then that just the same problem.

antoinevg, 2024-08-23:
>     2. The SoC which wraps the FPGA, can not get the bytes out of the FPGA logic "fast" enough.

Technically speaking the SoC is also implemented inside the FPGA.

The reason it's currently the bottleneck is that the peripheral interface between the USB gateware and the SoC gateware doesn't have sufficient bandwidth due to an architectural limitation. Lifting this limitation is the focus of our current research.

>     3. This is something that is unique to the Cynthion board.

Yes and no. GreatFET has a similiar limitation which is tied to the speed at which its NXP LPC4330 microcontroller can transfer data from one DMA buffer to another.

> Questions:
> 
>     1. Are there currently any USB2.0 devices that can serve as a face dancer backend and don't have this limitation?

I haven't had a chance to evaluate it myself yet, but https://github.com/HydraDancer might be worth a look?

> 2. If I understand you correctly, it also means that if the mass storage is all implemented in the FPGA, then that theoretically should solve the problem, however, you still would have to contact [...]

A pure FPGA implementation of mass storage would have direct access to external memory and would not need the SoC. I reckon you could easily saturate the bus with that approach.

straithe, 2024-09-19:
I'm closing this issue as it appears to be answered sufficiently. Please re-open this issue or open a new one if you still need assistance.

closed 2024-09-19 by straithe