greatscottgadgets/facedancer #111: facedancer.errors.DeviceNotFoundError: Could not find device to proxy

rlewkowicz, 2024-08-23:
I've used facedancer with rawgadget (hes got a custom branch) and I know it works. It finds my product and device id for an xbox one controller. 

I've tried linux and windows now just to see if it matters, and it does not. 

```
ID_VENDOR=0x045e
ID_PRODUCT=0x0b12
```

Is there a command to make the cynthion dump connected devices?

antoinevg, 2024-08-23:
Can you please confirm that your hardware connections match that in the following diagram:

https://cynthion.readthedocs.io/en/latest/getting_started_usbproxy.html

Thank you!

rlewkowicz, 2024-08-23:
Heyo! That was it, sorry I over looked that diagram and I believe earlier diagrams had the cynthion as the central unit.

I also had to do `BACKEND=cynthion` which I think is said somewhere in some docs, but the code has it as moondancer and I was trying to export that. 

So this just uses libusb then? I think that's the problem I'm having with rawgadgets as well then. I guess I thought this was going to be a magic bullet for mitm. On windows I dont think it can unbind? `usb1.USBErrorNotSupported: LIBUSB_ERROR_NOT_SUPPORTED [-12]`

Ultimately, proxying any modern game controller via anything is a pia. I'm trying to understand their auth handshakes, and I think this is where rawgadget fails too. It could be I'm on a pi too. 

Yeah, I just tried the analyzer with packetry and it works great. So I think libusb is the limitation. I'll set up a new x64 box and see if I can make that work. 

Maybe I'll stick with analyzer. It can do replays at a glance at the packety source. With my engineering goals I can probably make it do what I need without full mitm.

antoinevg, 2024-08-23:
> Heyo! That was it, sorry I over looked that diagram and I believe earlier diagrams had the cynthion as the central unit.

I made exactly the same mistake when I first started adding USB Proxy for Cynthion so don't feel too bad! 😂

> So this just uses libusb then? I think that's the problem I'm having with rawgadgets as well then. I guess I thought this was going to be a magic bullet for mitm. 

I _feel_ you. We are doing active research though towards moving the proxy function to Cynthion gateware in the medium term. It would actually help that work tremendously if you could please tell me a bit more about the kind of properties you'd like a magic bullet to have?

> On windows I dont think it can unbind? `usb1.USBErrorNotSupported: LIBUSB_ERROR_NOT_SUPPORTED [-12]`

Windows is a pain. Once the kernel has it, it's gone. You may be able to work around this by using Zadig to claim the device but this is very much uncharted territory at the moment.

> Maybe I'll stick with analyzer. It can do replays at a glance at the packety source. With my engineering goals I can probably make it do what I need without full mitm.

🚀

rlewkowicz, 2024-08-23:
I'm building, with some success, device manipulators like cronus or xim. I do this now in c++, but the goal is pick an ep, pick a direction, modify a bytestream, ship it. That would be my magic bullet mitm. I want to modif ep 84 on ps5 and I think 82 on xb1. Everything else should just work. 

The ps5 controller is going to get hairy. Even the big bois switched to remote play. On that one I think it's the isync channel. They're running a back channel with a hash or something and it blocks it if you modify the byte stream, but only on the actual console. I'm going to time couple the streams and train starcoder on it. starcoder is more "fill in the blank" than gpt conversation assistant etc. You can only do so much in so many bytes. They probably have some clever encryption but I wonder if you get enough tokens can you find some commonality? Then I'll have to modify that stream inline with the core input stream.

Then all this depends on if I have to get a job 😅. I got laid off and I'm just having fun but I'm running out of time. I gotta be employed soon unfortunately.

rlewkowicz, 2024-08-23:
More or less this is what Im building:
https://hireryan.today/2023/01/02/just-a-better-gaming-chair-honestly-destiny-2-aim-bot-using-yolov5/

It's come so much further than this and if I can get these devices down, I move to kickstarter and try to launch a company

rlewkowicz, 2024-08-25:
@antoinevg I just realized you wrote half/all of this. You have to be an interesting person this is all brilliant. 

In anycase, this exactly is the answer to #2. The magic bullet is bytes in bytes out. I feel like it's all much simpler in raw bytes: 
https://github.com/rlewkowicz/cythion-exploration/blob/826174a4fe472fedf94a7f4c5be8abd572a4e5a7/src/main.rs#L63

The code above will attach to the first cynthion board running analyzer, look for "4b20" which is a unique byte stream modify it print it. 

Can I replay this via these same packetry interfaces? I'm not familiar with rust and I'm still digging through the source, tests, etc.

Is the analyzer firmware source out there? Things are named interestingly sometimes.

Then finally, lets say this is my magic bullet. My background is secure software on linux. So I was going to bring that to a consumer device. Even if I need to build it myself, and it's just me and some raspberry pis, I can still make it so joe shmoe cant get in, and I run strong software interfaces, and build some number of units. 

But with this lets say I want to build a product. Yeah, the gateware, fabs, everything I can think of seems open source. But is that easier than trouble shooting libusb? Is there consultancy if these components are useful and I want to scale?

martinling, 2024-08-25:
> Can I replay this via these same packetry interfaces? I'm not familiar with rust and I'm still digging through the source, tests, etc.

No, you can't do any MITM/replay using Packetry and the analyzer gateware. When Cynthion is in use as an analyzer, the FPGA isn't participating in the exchanges being captured. The data lines of the Target-C port and Target-A port are hardwired to each other. All we do is sniff the traffic as it passes by, using the Target PHY in passive mode.

> Is the analyzer firmware source out there?

The analyzer is pure Amaranth gateware, there's no SoC or firmware involved. Source is here:

https://github.com/greatscottgadgets/cynthion/tree/main/cynthion/python/src/gateware/analyzer

closed 2024-09-19 by straithe