greatscottgadgets/facedancer #118: Facedancer --suggest mode doesn't print suggestions, as Ctrl-C is no longer caught

miek, 2024-09-19:
When running facedancer examples with the `--suggest` argument, pressing Ctrl-C to end the script just prints a backtrace instead of printing the aggregated suggestions. It used to catch the exception, but that got removed in https://github.com/greatscottgadgets/facedancer/pull/92

mossmann, 2024-09-25:
@kauwua Do you have any objection to reverting ea8c2a6?

antoinevg, 2024-09-26:
How about:

https://github.com/greatscottgadgets/facedancer/pull/119

kauwua, 2024-09-26:
catching `KeyboardInterrupt` would be ok in `default_main` since it's a convenience function (the main issue was Facedancer catching Ctrl-C directly in the core library and not exiting). this way you could print the suggestions, exit cleanly (with `exit` probably) to remove the ugly backtrace and users could create their own `main` if they want.

https://github.com/greatscottgadgets/facedancer/pull/119 PR works for me as well

miek, 2024-09-27:
I'm reluctant to introduce a new keyboard shortcut if we don't need to, I like the suggestion of catching `KeyboardInterrupt` in `default_main` instead.

mossmann, 2024-09-30:
@antoinevg plans to rework #119 so that the interrupt is caught in `default_main` and the `ctrl-e` solution is moved to an example.

antoinevg, 2024-09-30:
PR updated, feedback appreciated plz! #119

PR #119 (2024-10-08): Allow Control-C to gracefully exit a Facedancer emulation
This PR attempts to resolve the tension between:

- Being able to exit an emulation in order for a) the `--suggest` argument to work and b) just having something nicer than crashing out of the emulation with a `SIGINT` versus…
- Not interfering with any external scripts running one or more emulations.

~~I propose that we dedicate an unused key-combination (at the time of writing this is `Control-E`) that can be polled for in an `exit_handler()` co-routine.~~

The final approach taken is to catch the `KeyboardInterrupt` exception in the [... cut]
closed 2024-10-08 by mossmann