greatscottgadgets/facedancer #133: [3.0.5] Redesign of Descriptor handling broke Keyboard / rubber-ducky isn't working

jalmeroth, 2024-11-25:
Hi there,

in the latest release 3.0.5 (released today) there were some changes to the descriptor handling.
Unfortunately, this seems to break the Keyboard device at least.

Please find a full capture of the rubber-ducky example attached.

Best regards,
Jan
[facedancer.pcap.zip](https://github.com/user-attachments/files/17908893/facedancer.pcap.zip)

antoinevg, 2024-11-26:
Hi Jan,

1. Did you also do a `cynthion update` and a `pip install --upgrade facedancer` after updating to the latest cynthion package?
2. Can you please also let me know which operating system(s) + version(s) you are running on?

Thank you!

jalmeroth, 2024-11-26:
Hi Antoine,

you're right, I should have included this info in my initial report. Sorry about that:

> 1. Did you also do a `cynthion update` and a `pip install --upgrade facedancer` after updating to the latest cynthion package?

Yes, I did upgrade both cynthion devices upfront:

```
> cynthion info
Cynthion version: 0.1.8
Apollo version: 1.1.0
Python version: 3.12.7 (main, Oct  1 2024, 11:15:50) [GCC 14.2.1 20240910]
```
This is how my python venv looks like:
```
> pip list freeze
Package                       Version
----------------------------- --------
amaranth                      0.4.1
apollo-fpga                   1.1.0
backports.functools-lru-cache 2.0.0
construct                     2.10.70
cynthion                      0.1.8
deprecation                   2.1.0
facedancer                    3.0.5
future                        1.0.0
Jinja2                        3.1.4
libusb1                       3.1.0
luna-soc                      0.2.0
luna-usb                      0.1.2
MarkupSafe                    3.0.2
packaging                     24.1
pip                           24.3.1
prompt_toolkit                3.0.48
pyfwup                        0.5.2
pygreat                       2024.0.2
pyserial                      3.5
pyusb                         1.2.1
pyvcd                         0.4.0
pyxdg                         0.28
tabulate                      0.9.0
tomli                         2.0.2
tqdm                          4.66.5
usb-protocol                  0.9.1
wcwidth                       0.2.13
```

> 2. Can you please also let me know which operating system(s) + version(s) you are running on?

I'm running facedancer on Arch Linux:
> Linux 6.11.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 17 Nov 2024 16:06:17 +0000 x86_64 GNU/Linux

Greetings,
Jan

antoinevg, 2024-11-26:
Tx for the reportage Jan, I'll be pushing out a new release with the fix this week.

jalmeroth, 2024-11-26:
Awesome, thanks Antoine!

I can confirm, that the proposed changes in #134 are fixing the issue for me. :+1:

PR #134 (2024-11-27): Get rubber-ducky example working with new descriptor features.
This PR does a bunch of things:

* There is some ancient confusion in bits of the codebase when referring to a descriptor's `number` and `type_number`. The `number` is the descriptor _number_ whereas `type_number` is the descriptor _type_. This was causing identifier lookups to fail when searching for a descriptor in `requestable_descriptors`.
* The `Keyboard` device's `HIDDescriptor` also needed to specify `include_in_config`.
* The data class for `USBClassDescriptor` needed a type annotation.

---

closes #133
closed 2024-11-27 by antoinevg