greatscottgadgets/facedancer #142: Questions about proxy.py

mwuhang, 2024-12-12:
Hardware: facedancer
Software: kali, python3.8.9

After I downloaded the file from the library, I tried to run the facedancer\proxy.py file, but it always reported a dependency library problem

I am a newbie, can you please give me some guidance?

If this hardware/software doesn't work, is there any other way to enable the proxy? What I want is to capture the data packets fed back by the device under test

antoinevg, 2024-12-12:
Hi, thank you for your questions!

Can you please copy and paste the error message that's being reported?

mwuhang, 2024-12-12:
At first, I downloaded version 3.0.6 of the library locally and tried to run the proxy.py file. I encountered the first problem as follows:
Traceback (most recent call last):
  File "/xxx/facedancer/facedancer/proxy.py", line 11, in <module>
    from .           import DeviceSpeed, USBConfiguration, USBDirection
ImportError: attempted relative import with no known parent package

For example, the import command from .request import USBControlRequest in the file, I deleted all the "." in the import command, renamed the error-prone types.py and logging.py files, and updated them to the various files in the facedancer folder.
Later, because the configuration.py file and the device.py file were imported in a loop, an error occurred. I merged the previous two files into con_device.py and made appropriate modifications after the merger. The current error content is:

Traceback (most recent call last):
  File "/xxx/facedancer/facedancer/proxy.py", line 13, in <module>
    from con_device     import USBBaseDevice, USBConfiguration
  File "/xxx/facedancer/facedancer/con_device.py", line 33, in <module>
    @dataclass
NameError: name 'dataclass' is not defined

That's my question. Thank you for your patience.

antoinevg, 2024-12-12:
The main problems you're running into are:

1.  You need to first install the Facedancer library into your local Python environment. See https://facedancer.readthedocs.io/en/latest/getting_started.html for more information.
2. If you want to run library files directly you'll want to specify the module rather than the path. i.e. `python -m facedancer.proxy`
3. There's also a standalone example in the `examples/` directory you can run with `python ./examples/usbproxy.py`.

Can you please also tell me more about the hardware you're using, you said it is 'facedancer' but there are many different Facedancer devices such as the GoodFET, Cynthion, Hydradancer etc.

mwuhang, 2024-12-13:
Thank you very much for your help. My facedancer is a GoodFET device. I found my problem. When I called python ./examples/usbproxy.py, my python could not recognize the device. I also tried many methods (for example: export BACKEND=goodfet; try to change the facedancer library file, etc.). I also referred to the web tutorial you mentioned. Finally, I found that there was a problem with my hardware device. When I ran python ./examples/usbproxy.py, my kali could not recognize GoodFET's facedancer. However, when I tried to run the file https://github.com/travisgoodspeed/goodfet, there was a client/ in its directory. When I tried to call python facedancer-keyboard.py, there was no problem of not being able to recognize facedaner. It always kept running normally. It may be that my opening method is different. I think I may need to try a new device again. Thank you again for your patience. Thank you very much.

jfshza303, 2025-01-19:
Did you ever get this issue resolved and if so please provide the steps.