greatscottgadgets/facedancer #148: Problems with proxying devices with cynthion

dumitory-dev, 2025-01-02:
Hello!

I am having trouble proxying devices, I am using such a 

usb flash -> host -> cynthion -> target 

When I use the proxy example  I get the following output:

```
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [13:36:14] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=64, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | logging        | [13:36:14] <: b'\x12\x01\x10\x02\x00\x00\x00@Q\t\xae\x16\x10\x01\x01\x02\x03\x01'
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [13:36:14] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=18, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | logging        | [13:36:14] <: b'\x12\x01\x10\x02\x00\x00\x00@Q\t\xae\x16\x10\x01\x01\x02\x03\x01'
INFO    | logging        | [13:36:14] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=771, index=1033, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | logging        | [13:36:14] <: ̲C87F54C797A7E890A9070443
INFO    | moondancer     | Target host configuration complete.
INFO    | logging        | [13:36:14] USBControlRequest(direction=0, type=0, recipient=0, number=9, value=1, index=0, length=0, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [13:36:34] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=64, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | logging        | [13:36:34] <: b'\x12\x01\x10\x02\x00\x00\x00@Q\t\xae\x16\x10\x01\x01\x02\x03\x01'
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [13:36:34] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=18, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | logging        | [13:36:34] <: b'\x12\x01\x10\x02\x00\x00\x00@Q\t\xae\x16\x10\x01\x01\x02\x03\x01'
INFO    | logging        | [13:36:34] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=771, index=1033, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7e3f0138a5f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7e3f003c8310>))
INFO    | logging        | [13:36:34] <: ̲C87F54C797A7E890A9070443
INFO    | moondancer     | Target host configuration complete.
INFO    | logging        | [13:36:34] USBControlRequest(direction=0, type=0, recipient=0, number=9, value=1, index=0, length=0, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: [... cut ...] ost_init__.<locals>.<lambda> at 0x72d2cea865f0>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x72d2cda8c310>))
INFO    | logging        | [13:47:05] <: ̲C87F54C797A7E890A9070443
INFO    | moondancer     | Target host configuration complete.
```

What can I do? Thank you

dumitory-dev, 2025-01-02:
I tested another devices

it seems that works correctly only if connected device speed 12mb (full speed)

I think it's should be works with high-speed mode, but only full speed now working

dumitory-dev, 2025-01-02:
There is logs if i connect any device with high-speed mode

```
INFO    | __init__       | Starting emulation, press 'Control-C' to disconnect and exit.
INFO    | moondancer     | Using the Moondancer backend.
INFO    | moondancer     | Connected HIGH speed device 'facedancer.proxy.USBProxyDevice' to target host.
INFO    | device         | Host issued a bus reset; resetting our connection.
...

```

antoinevg, 2025-01-06:
Could you please provide some more information about the operating system(s) you are  running on your host and target machines?

H1X4Dev, 2025-01-09:
I tried it too, but it doesn't seem like it works when Windows 11 23H2 is used as the host.

Upon setting up a VM (Ubuntu Server 24.04) and using the passthrough for USB mouse and Cynthion device and then use the same script, it works with an issue of very noticeable input lag and Cynthion itself gets very warm. Using the windows as target works fine however.