greatscottgadgets/facedancer #152: USBConfiguration.from_binary_descriptor() fails for FT600 Function with InterfaceAssociationDescriptor

mblsha, 2025-02-02:
Packetry shows the following device tree for an FT600 device:
[image]

Here's the binary for the "Getting configuration descriptor #0 for device 1, reading 63 bytes":
```
b'\t\x02?\x00\x02\x01\x00\xe00\x08\x0b\x00\x02\xff\xff\xff\x00\t\x04\x00\x00\x02\xff\xff\xff\x00\x07\x05\x01\x02\x00\x02\x00\x07\x05\x81\x03@\x00\t\t\x04\x01\x00\x02\xff\xff\xff\x00\x07\x05\x02\x02\x00\x02\x00\x07\x05\x82\x02\x00\x02\x00'
```

And calling `USBConfiguration.from_binary_descriptor()` results in an error:

```
  File "/Users/mblsha/.pyenv/versions/3.12.5/envs/ft600/lib/python3.12/site-packages/facedancer/configuration.py", line 95, in from_binary_descriptor
    if len(last_interface.endpoints) == 0:
           ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'endpoints'
```

[configuration.py](https://github.com/greatscottgadgets/facedancer/blob/main/facedancer/configuration.py#L94-L98) expects there to be either an interface or an endpoint before a USBDescriptor, but in this case USBDescriptor is the first thing that gets parsed in that loop.

[In packetry](https://github.com/greatscottgadgets/packetry/blob/12a867bfc082a1712e99b5de00d0a644fa0a0df7/src/usb.rs#L750-L779) this seems to be parsed as a Function with a InterfaceAssociationDescriptor, but there seem to be no such thing in facedancer.

Also not sure if it's important, but the configuration description is getting transferred twice, and the second one seems to be a superset of the first:
[image]

mblsha, 2025-02-02:
Hmm, connecting it to Linux makes the kernel complain about lack of BOS descriptors, which facedancer also doesn't seem to support. And this might be related to the USB3 support in general.

I connected FT600 to host using a simple USB2 cable to allow Cynthion to sniff the data, and in general making it work for USB3 devices but at slower speed would be a nice thing to have, but might be outside of the current scope of Facedancer.

martinling, 2025-02-02:
> [configuration.py](https://github.com/greatscottgadgets/facedancer/blob/main/facedancer/configuration.py#L94-L98) expects there to be either an interface or an endpoint before a USBDescriptor, but in this case USBDescriptor is the first thing that gets parsed in that loop.

Yes, this is the problem exactly. The configuration parsing assumes that descriptors embedded in the configuration data will only be attached to interfaces and endpoints. Interface association descriptors break that pattern, causing the failure.

The fix is going to be either to implement specific support for interface association descriptors in Facedancer, or at least to support generic `attached_ descriptors` for configurations, as we do for interfaces and endpoints.

> Also not sure if it's important, but the configuration description is getting transferred twice, and the second one seems to be a superset of the first:

This is normal (but OS-dependent). It happens because the host requests the configuration descriptor itself first (9 bytes), then checks the `wTotalLength` field and makes a second request for that full length.

> Hmm, connecting it to Linux makes the kernel complain about lack of BOS descriptors, which facedancer also doesn't seem to support.

Facedancer supports arbitrary descriptors even if it doesn't understand them. For your BOS descriptors you should be able to include something at the `Device` level like:

```python
@requestable(type_number=0x0F, number=0)
class BOSDescriptor(USBDescriptor):
    raw = # insert your raw descriptor bytes here
```

mblsha, 2025-02-03:
Thanks Martin! 

> The fix is going to be either to implement specific support for interface association descriptors in Facedancer, or at least to support generic attached_descriptors for configurations, as we do for interfaces and endpoints.

Should I leave the issue open if I don't plan to do it myself? (my initial goal to understand enough of FT600 to make it work with WebUSB, but I'll probably continue using the Python library for now since it should be good enough for my use-case)

martinling, 2025-02-03:
Yes, it's a bug and we should fix it.