greatscottgadgets/facedancer #159: Facedancer and Cynthion, how to add more "strings" to the USB Descriptor (Getting string descriptor #9)?

PaneDmitrovic, 2025-05-22:
Hi again everyone,

Unfortunately I have an "interesting" problem again.

Namely, my device has 9 strings in the DESCRIPTOR as you can see from the PACKETRY analysis:

`Getting string descriptor #9, language 0x0409 (English/US) for device 2, reading 36 of 255 requested bytes: 'AT9*v1.20' "
`

[image]

As you can see the request for my strings are classic descriptor requests:

> SETUP transaction on device 2, endpoint 0 with 8 data bytes, ACK response
> Payload: [80, 06, 09, 03, 09, 04, FF, 00]

Of course the first three requests are successful because they are:

1. manufacturer_string
2. product_string
3. serial_number_string. 

Although the original device does not have a serial number listed, when I enter "string descriptor #3" in "serial_number_string" the emulation works normally until the next string is requested.

If you need more information feel free to contact me.

antoinevg, 2025-05-23:
Can you give this a try?

```python
class MyDevice(USBDevice):
    ...
    product_string       : str = "My Device"
    manufacturer_string  : str = "Facedancer"
    serial_number_string : str = "myserial"

    def __post_init__(self):
        super().__post_init__()

        self.strings.add_string("my string", index=index_for_string)

    ...
```

PaneDmitrovic, 2025-05-24:
Works perfectly, thank you very much @antoinevg 👍

closed 2025-05-24 by PaneDmitrovic